system76/laptop-suggestions #10: European language keyboard options.

emiljoha, 2019-11-21:
### Why/User Benefit/User Problem
Many European languages have letters not present in the English and thus different keyboard layouts.

The keyboard layout is ingrained in muscle memmory and stickers is not the same. Not having a language correct keyboard will scare off large portions of customers. Me included. 

There are many considerations when buying a new laptop, many are small, this is foundational for reaching European markets. 

### Description of the feature

Add more ( e.g Swedish ;) ) keyboard layout options.

fleskesvor, 2019-11-22:
The Scandinavian keyboards are similar enough that some manufacturers provide the same keyboard with side by side alternative markings. I think that's a good solution, though I'd prefer a Norwegian keyboard, if they are custom made per order.

muhlinux, 2019-11-23:
Nordic Keyboard :+1:

phma, 2019-11-29:
I touch-type Dvořák. Can I get a Dvořák keyboard?

I also switch to Greek, Hebrew, or Russian layouts when I need to type in those layouts, but I don't have them completely memorized (especially the Hebrew). Being able to select the layout when buying a laptop would be good.

oskarcafe, 2020-01-07:
I own a System 76 laptop, and I mix coding and writing text (many å, ä ö:s...), so I'm not ready to start using US layout even if it is technically better for coding. Now - it works with Swedish stickers, but since the US layout lacks one key - compared to the UK/European one - it doesn't work. Since it's pipe, < and > that are missing, crucial for any coder, one more key is a dealbreaker for me.

gregoryloichot, 2020-08-04:
Yes, please please allow us to select other keyboard layouts.
Swiss French would be appreciated :)

qwattash, 2020-12-24:
To be honest, I don't care about the language but at least having an option for an ISO keyboard instead of ANSI would be a priority for me.

Rudde, 2024-04-10:
Any update on this?

leviport, 2024-04-10:
We offered an ISO layout briefly on one model a few years ago, just to test demand. They didn't move quickly.

That said, we eventually want to set up a distributor in Europe, wherein I'm sure ISO layout keyboards would be in much higher demand. There is no timeframe on this venture, but we've done a fair bit of research and planning to get ready for when this becomes feasible.