system76/laptop-suggestions #12: 3:2 Screen like Surface Laptop or Huawei and thin bezels

jerrmo, 2019-11-21:
### Why/User Benefit/User Problem
In my experience, the 3:2 screens make for a better browsing/productivity form factor. Also, thin bezels make for a sleeker, more modern look. The chunky "chins" on most modern 16:9 laptops (HP, Dell, Lenovo) look outdated.

### Description of the feature
3:2 Screen Ratio like the Surface Laptop or Huawei Matebook X Pro, with thin bezels. The Surface Laptop is a very good example of a screen with thin bezels and 3:2 aspect ratio while also keeping the webcam where it is supposed to be (at the top, not at the bottom or in a weird button on the keyboard)

ahoneybun, 2019-11-21:
I think it's not a good aspect resolution. 16:9 is a good area for laptops to use and it's good for media consumption.

HansJK, 2019-11-21:
I think 16:10 just like the MacBooks have, is the perfect screen ratio. It makes for a great browser experience, but also works fine having two windows side by side. It also looks very modern.

Venturub, 2019-11-21:
I prefer 3:2 but, regardless, thin bezels for sure!

mtkunkel, 2019-11-22:
Anything taller than 16:9 please.   Yes, the screens will be more $$$.   Go against the typical consumer trash mass produced parts and make something great.   16:10, 3:2, or 4:3..   Any are better

jerrmo, 2019-11-22:
I agree - 16:10 is fine but 3:2 is preferred. Please no chunky "chins" on the bottom. Lets get those thin bezels!

NOTtheMessiah, 2019-11-23:
3:2≅1.5 is the best for productivity on a tablet. For dual pages, it approximates the (2*8.5)/11=17/11≅1.54 ratio for two pages side by side, is relatively good for single page viewing 8.5:11≅1.29 (using up 86% of screen space), isn't too far from 16:10≅1.6 ratio which is probably best for all-purpose use (and notably the choice for MacBooks for good reason) or 16:9≅1.77 for media consumption (using up %84 of screen space for video). No use case above drops below 84% of screen usage.

16:10 uses 90% of screen space for media consumption, yet is also very good for documents side-to-side, preferable laptops with larger screens.

I should organize this into a table...

ajflores1604, 2019-11-25:
+1 for 16:10 or 3:2 screen. It's incredibly frustrating programming on a 16:9 screen after coming from something taller