system76/laptop-suggestions #14: AMD option targeting developers, content creators, and power users

cmack, 2019-11-21:
### Why/User Benefit/User Problem
It's hard to find laptop makers that pair premium features with AMD CPUs. For example, models with high resolution screens and large memory and storage capacity are not usually provided with AMD CPU-based SKUs. Or they are only available with "gaming" segment models that are usually bulkier and heavier.

The high-core-count AMD CPUs paired with premium features would appeal to developers and content creators. 

### Description of the feature
1. Provide an option for latest generation AMD CPUs 
2. Give the same high grade options that other SKUs using different CPU manufacturers receive. For example, if the other SKU's highest grade config allows for 4K monitor, 1TB storage, and 16+ GB of RAM, the AMD SKU should have those options too.

kolgza, 2019-11-21:
What would be **really** interesting is if the [AMD PSP ](https://en.wikipedia.org/wiki/AMD_Platform_Security_Processor)could be disabled the same way the [Intel Management Engine was](https://liliputing.com/2017/11/system76-will-disable-intel-management-engine-linux-laptops.html).

AlNotSteve, 2019-11-21:
Up vote this!

rakshith-ravi, 2019-11-22:
An AMD system, with a small form factor (13-14 inch), comfortable, no-numpad keyboard (like the ThinkPads) with high specs (enough to game 60+ fps with an external monitor) would be the most ideal system of all time. Can we please make this happen @system76?

barrel0luck, 2019-11-22:
Yes! Yes! Yes! Given all the security nightmares with Intel chips, AMD option will be much appreciated!

mtkunkel, 2019-11-22:
Is AMD + Thunderbolt 3 do-able yet?   Potentially a first in the market..   (one can dream)

victorolinasc, 2019-11-28:
Fully performant GPU with mainline kernel only...

No graphics driver blob...

No kernel update hell... 

I dare you System 76.... I double dare you...

Owslla, 2020-01-02:
Up voting this suggestion.