system76/laptop-suggestions #15: High Dpi IPS screens with a glossy option.

HansJK, 2019-11-21:
High quality IPS display with an option for a glossy finish in order to get more vibrant colors and deeper blacks. Helps people who works with design. And of course reduce the glare like apple have done on their displays. 16:10 would also be ideal, giving a richer browsing experience, more modern look, while still being able to have two windows open at the same time.

julienbenjamin, 2019-11-23:
Glossy means reflections. Who likes that? Especially on a laptop...

HansJK, 2019-11-24:
> Glossy means reflections. Who likes that? Especially on a laptop...

I said as an option because I know people like you exist out there. If you look at high quality displays out there, there's very little reflection these days, compared to cheap ones. And when you get glare, it'a usually only at a certain spot on the display, while a mat display spread the glare across the whole surface. These means that a glossy display still stays ultra crisp and clear for about 80 - 90 % of the display while only being blurred at a certain spot. But a mat display ends up getting about 10 - 20 % les crisp and clear across the whole surface. that's one of the reasons why people who work with creative work tend to like glossy displays.

julienbenjamin, 2019-11-24:
Long story short, no, this is because panel such as VAs and derivatives get a significant boost in contrast for example.

Usually, "creatives" like you said, do not really work on laptops. And glossy screens on laptops appeared to increase battery life and, again, show a better contrast ratio.

People like me, like you said, tend to be willing to be able to read/see what's on their screen before anything else, even "creatives one". Because, even with recent ones and contrary to what you say, glossy panels do not let you see (and even less with fidelity) what's on your screen when their is natural ambient light.

jpkx1984, 2023-03-15:
> Long story short, no, this is because panel such as VAs and derivatives get a significant boost in contrast for example.
>
> Usually, "creatives" like you said, do not really work on laptops. And [...]

But that's you. I like glossy laptop screens more I use them for software development when on the go (at home I use a desktop). The UI looks more vibrant to me on glossy. Unless in direct sunlight (which sucks regardless of the screen used), reflections are not a big deal.