system76/laptop-suggestions #19: 3k Bezeless touchscreen

Venturub, 2019-11-21:
Gnome, the DE used in PopOS, has a design thought to be used in tablets. It is even now phone-ready with Librem5. Yet few, to no Linux laptops offer touchscreens.

Also, few to no Linux PC offers a ultra high definition screens, or 3k screens, not even as an option. That may be important for many video and photos professionals... if not even also for gamers...

peircej, 2020-02-04:
For me lack of touchscreens is one of the things that holds me back from linux laptops

forrestbao, 2024-02-03:
I really want to see a touchscreen laptop from System76. I really like to draw on the touchscreen when I collaborate with teammates in other parts of the world.