system76/laptop-suggestions #20: Thin, light weight, integrated graphics with good battery and quiet operation

madredhatter22, 2019-11-21:
### Why/User Benefit/User Problem

### Description of the feature

murmrs, 2019-11-22:
Cool with thin and light, but that doesn't usually equate good battery, especially on Linux. I have an X1C now, something in that size range (and it doesn't need to be built of aluminum to _feel_ good), would probably be good.

TeamLinux01, 2019-11-22:
If at all possible, I would like to see a completely fanless model with great battery life.

victortrac, 2019-11-22:
I came here to say this. My favorite laptop of all time is the X1 Carbon. If system76 can make it as close to the X1C as possible (14" screen in a 13" body, light, decent screen, 8+ hr battery life), plus have it support 32GB of RAM, it'd be the ultimate laptop.

tmtvl, 2023-04-05:
Thin is worthless.  It's better to have a laptop that works well and checks all the necessary boxes over one that is very thin but crap.