system76/laptop-suggestions #3: No webcam as option

neuromancer2049, 2019-11-21:
Please allow an option to have no built in camera. Why?

More screen space to work with. This would open up the abilities of added personal security (without the tape over the camera ruining the beauty of your system76 products) and grant additional screen space and possibly edge to edge design. This would be useful in all your laptop configuration. Especially the workstations.  (There are options for webcam shutters , however they aren't worth the hassle. And break easily)

nuk380y, 2019-11-21:
Second this.  Either an option to not have a webcam _at all_, or the lid has a built-in camera slide/cover.

miharkula, 2019-11-21:
Agreed. I'd rather have more screen space/less bezel than a webcam

mattwhiteside, 2019-11-26:
+1 to this.  I'd rather see engineering effort put elsewhere than the webcam, which I never use.