system76/laptop-suggestions #5: Battery life > resolution/touch

wmww, 2019-11-21:
### Why/User Benefit/User Problem
High resolution screens and touchscreens suck a lot of power.

### Description of the feature
While some users might want >2k resolution and a touchscreen, I think many would prefer longer battery life. There should at least be an option for that.

emiljoha, 2019-11-21:
In addition it will probably be a good old while before high DPI screens are actually a better Linux experience.

anilsg, 2019-12-26:
I didn't +1 because the title seems to be implying the reverse of what I'm hoping this means, but I'm commenting here to support high battery life. I don't want a touch screen or v high resolution. I'm currently happy with X1 14" 1920px and I don't want touch screen or v high resolution at the cost of power drain (and $). I don't get why touch screen is such a thing for laptops now. If I want a tablet I'll get one.

enetics-secker, 2019-12-26:
...and USB-C as the primary charging interface.

briggs1978, 2020-04-10:
High PPI screens are a nightmare for me personally, so I will always choose battery life over high res.