system76/laptop-suggestions #6: Convertible or 2-in-1 laptop

ahmed-elsayed2017, 2019-11-21:
I would like on-the-go laptop that is light weight that can be deattached to function as a tablet, or a convertible laptop that can be rotated 360 degree, but it must be light weight, and the CPU must be AMD Rayzen, because Rayzen series have a powerful iGPU that can run video games.

miharkula, 2019-11-21:
I second this but world like to add including auto disabling keyboard if flipped over (for 360 option)

anorman728, 2019-11-21:
I was going to post the same thing.  I'd also add that I prefer 10 inch to 13 inch.  I have a SurfaceBook, but I never use it as a tablet because it's too large to hold.

wmww, 2019-11-21:
If you make a convertible, please don't disable the keyboard in hardware when flipped. Connecting to a projector, placing it screen-down and using the laptop as just a keyboard is a valid usecase. I've had a laptop that made this impossible because the keyboard was disabled in hardware when the screen was flipped.

ahmed-elsayed2017, 2019-11-22:
> If you make a convertible, please don't disable the keyboard in hardware when flipped. Connecting to a projector, placing it screen-down and using the laptop as just a keyboard is a valid usecase. [...]

It is unpopular opinion, but if is there an option available inside BIOS to turned it off if you want. That will be good for people who doesn't want the keyboard to be disabled.

ApatheticCosmos, 2019-12-10:
I'm typing this on a first gen Galago, and I'm looking to finally upgrade. This laptop has served me very, very well, but it's getting old and kind of beat up.
I'm conflicted. I want to support System76 because they put so much effort into polishing the Linux experience.
However I also really want a convertible tablet.

I found this issue by searching Google for "System76 Convertable".
P.S. I'd prefer an AMD apu... but I guess it's not a dealbreaker.

NiklasBeierl, 2020-07-28:
Just here to say that I would immediately order two of those if it features touchscreen and a **stylus**.
Ideally:
13 inch
2K display (2K is the sweet spot for that size imho, sharp enough to make the pixels on Xournal annotations disappear)
Long battery life
Bluetooth 
Thunderbolt 3 for docking stations. 

Written on an XPS 13 convertible with driver issues... :disappointed:

KorraHarraway, 2021-01-21:
A 2-in-1 linux macine is a must overall. In my case, a Lenovo Yoga have been my production machine for 5+ years right now. I guess a native linux 2-in-1 machine would be more targeted toward industry design professionals, teachers, professors, and researches, since the touch and stylus capability with full linux support has not that broad audience among creative professionals and media consumers. Devs and IT workers in general choose to stick with clamshells unfortunately. Where 2-in-1 have been absolutely very popular is among business and corporate people, but they tend to stick with Windows in any event. 

In the direction of a 2-in-1 workstation, the best Windows 10 convertible implementation I have seen so far is the Acer Conceptd 7 Ezel. But Windows is not an option at all.

Thus, for a 2-in-1 native linux machine, I think a larger OLED display with high color accuracy would be ideally better. A UHD+ 15" with a 35+ watt processor paired with a discrete GPU seems to be a sweet spot. This is because a small 13" or 14" display without discrete GPU and with a lower powered CPU would conflict with for example the latest Thinkpad X1 Yoga which can be shipped with native linux support as of now.

As for the touch and pen input, in my experience Wacom based displays use to have better linux support, but I'm not sure how N-trig based displays have been working with linux these days, however it seems most OLED 2-in-1 laptops in the market right now are using N-Trig.

An AMD Ryzen 5000 CPU would be better, but I understand that, until USB 4.0 get widespread, thunderbolt is necessary for some people.

youdontneedtoknow22, 2021-11-13:
I'm actually just a student at college and I would only use 2 in 1 laptops. I don't want to have a seperated tablet for studying (to make notes on lectures) and a laptop for reading and to take with me to college (because it has a bigger screen, I don't feel comfortable learning from a tablet).
So I didn't buy any desktop or a tablet, just an x360 model from HP (for 830€) with i5 8th gen. and 8 gigs of RAM and it's working good (on windows, speakers suck on linux).

Also, AMD isn't a must anymore. Intel has Intel iris xe which should even beat rx radeon. (UHD 620 still sucks for gaming tho)

pktiuk, 2024-01-05:
When I was looking for a proper 2in1 Linux laptop (with real GPU) I couldn't find anything with proper linux support.   
And finally I went with [Asus ROG X13 with RTX 3050Ti](https://rog.asus.com/laptops/rog-flow/rog-flow-x13-2022-series/). This is a decent machine, but lack of proper drivers in mainline is a real headache. (problems with hybrid GPU, battery, disabling keyboard in tablet mode etc). I would gladly pay more for machine which just works.

forrestbao, 2024-02-03:
I really want to see a touchscreen laptop from System76. I really like to draw on the touchscreen when I collaborate with teammates in other parts of the world.

joelsolo99, 2024-09-19:
I would also immediately buy a 2-in-1 from s76. Currently using a 3 year old envy x360 that would be perfect if it had more than 8gb RAM and had better driver support on linux. 2-in-1 allow me to use my laptop in tent mode as a monitor, with an external keyboard and mouse for a portable desktop experience. 13/14", 1440p, lots of USB-C, stylus support would be absolutely perfect.