system76/laptop-suggestions #8: Hardware switches for WiFi, Bluetooth, microphone, and webcam

nuk380y, 2019-11-21:
### Why/User Benefit/User Problem

Including hardware switches for HIDs allows the user to minutely control which devices are active or can be activated on demand while also ensuring that said devices cannot be activated without the user's knowing (assuming that the switches are in the "off" position).  Added benefit for the extra-cautious/paranoid crowd.

### Description of the feature

Hardware toggles above the keyboard (preferable) or on the side of the device that physically disconnect/disrupt connection of transmission and recording HIDs (WiFi, Bluetooth, microphone, webcam).

tdapple, 2019-11-21:
I second this one for sure.

AlNotSteve, 2019-11-21:
I add my second to this one!

Venturub, 2019-11-21:
Add mine also!

barrel0luck, 2019-11-22:
I must also advocate for kill switches for everything including hoomans.

davidallen353, 2019-11-23:
Kill switches would be great. An option to buy a unit without any cameras or microphones would also be useful.

bcopp, 2020-03-05:
Kill switches for sensitive components like the camera and microphone are an essential security feature. I would love to see this feature implemented.

G0rd0n, 2021-06-15:
In the meantime, I decided to tackle this issue (for the microphone only) by ; 
1) requesting the hardware schematics for my laptop from the S76 team.
2) find the webcam \ microphone connector and note down the corresponding name of the connector from the motherboard, look up the wiring schematics for the connector and find the microphone connection cable (MIC_DATA_R in my case).
3) taking my old iPhone apart to get the mute switch (lengthy process, but iFixit provides a good walkthrough), microsolder some new wires on the switch.
4) drill a hole in the outer case in a place where it makes sense, attach the switch. Cut the microphone data wire and connect it to the mute switch (to do).
5) lose all warranty. (mandatory)