system76/laptop-suggestions #9: USB C Charging

aymswick, 2019-11-21:
### Why/User Benefit/User Problem

### Description of the feature

codeallstar, 2019-11-21:
I second this. I would of bought a darter pro in the spring instead of a HP Elitebook. It would of saved me time wiping Win10 and loading Pop!_OS

RiderExMachina, 2019-11-22:
Maybe I'm in the minority, but I prefer barrel connector chargers.

rakshith-ravi, 2019-11-22:
> Maybe I'm in the minority, but I prefer barrel connector chargers.

I'm curious, why do you prefer barrel connectors? Maybe you can give us a different opinion that'd open our minds?

ptanmay143, 2019-11-22:
I second this. Having a single type of charger for both my phone and my laptop would be a huge win for me.

Carreau, 2019-11-24:
Two advantages of that on my current work laptop are: 
1) I now have a charger port both on left and right of my machine. 
2) You can daisy-chain charging which is useful for example at conferences where there are limited number of outlets.

tekkamanendless, 2019-11-28:
USB-C charging is my hard requirement for a new laptop (because I can use Volta XL cables for magnetic breakaway support). This would be huge for me.

RiderExMachina, 2019-12-10:
> I'm curious, why do you prefer barrel connectors? Maybe you can give us a different opinion that'd open our minds?

Because all barrel connectors will 100% be guaranteed to charge your laptop whereas USB C has already had issues due to different specs within the standard (namely the Switch, RasPi4, and Apple devices not working without their official chargers)

kusuriya, 2019-12-26:
> Because all barrel connectors will 100% be guaranteed to charge your laptop whereas USB C has already had issues due to different specs within the standard (namely the Switch, RasPi4, and Apple [...]

100% that's not even remotely true, I have plenty of barrel chargers that will sometimes charge due to issues in their, or the laptops manufacturing. Honestly Is have to get more data but I'm pretty sure the failure rate for USB-C and barrel chargers is near the same if you compare like to like, so smart barrels against USB-PD and dumb against dumb. Most of the issues you note are issues in manufacturing where people didn't follow the spec correctly and Smart chargers wig out

Carreau, 2020-01-31:
Note that European commission want to enforce a single charger type for most electronic devices:
https://edition.cnn.com/2020/01/30/tech/eu-apple-single-phone-chargers/index.html

That might want to be something to keep an eye on.

22bsti, 2020-03-19:
I agree USB-C even with the quirks is preferable to barrel plugs.

briggs1978, 2020-04-10:
One USB-C charging port on either side is an absolute requirement now for me to purchase a laptop.

neuromancer2049, 2020-04-12:
What do you think of the concept of "mirrored ports"?  What you got on left side is exactly same ports on right side. Can charge left or right. Without being too "awkward" somehow.

neuromancer2049, 2020-04-12:
> ### Why/User Benefit/User Problem
> One cable to rule them all - barrell connectors are bulky
> 
> ### Description of the feature
> Offer enough usb-c ports to charge the laptop and still have one free for data transfer.

But how much is needed ? Is 3 usb c too little ? Or is 4 a good number since they hardly take any space ?

tincho, 2021-08-11:
hello! are currently (august 2021) available any System76 laptops that charge over USB-C port? Specifically the Lemur model? Thanks!

leviport, 2021-08-11:
@tincho USB-C charging is listed in the Tech Specs section when it is supported: https://system76.com/laptops/lemur/

[image]

jonocodes, 2022-09-09:
Lets get this for the galago.

leviport, 2022-09-12:
The galp6 does support USB-C charging. It appears it was left off the tech-specs by mistake, but it is mentioned in the technical documentation: https://tech-docs.system76.com/models/galp6/README.html

We will correct the tech-specs very soon.

LukeClancy, 2023-10-12:
In my experience, I have a barrel plug connector on my gaze14 which failed, so I had to get it sent in. Now it is currently mid-failure again. USB-C is known to be relatively reliable, so I would like that solution in the future. Preferably not soldered onto a circuit board, so that it is easily replaceable. 

Every time something like this happens, it majorly impacts my work due to around 2 weeks of turnover when it gets sent in. I would not get a computer without usb-c charging in the future. Simply because I know what I am getting (instead of the shoddy component I have now).