system76/laptop-suggestions #22: Keyboards without numpads on high-end laptops

elpete, 2019-11-22:
### Why/User Benefit/User Problem
I prefer not to have a numpad on my keyboard. I rarely use it, and it causes the entire keyboard to shift to the left when it is included making me feel like I’m off center the entire time I’m using the laptop. 

### Description of the feature
A nice keyboard with no numpad, kind of like Apple laptops have (but working 😏), on a high-end laptop, not just the “ultrabooks”.

RiderExMachina, 2019-11-22:
Maybe give the option to not have a numpad as I use mine a lot and hate when my keyboard is missing a numpad.

thatguystone, 2019-11-23:
Having multiple options for keyboard layouts would be awesome. Personally, I really like the current layout of the Galago (with home/end/pgup/pgdown).

briggs1978, 2020-04-10:
Hooray for centered keyboard and touchpad.  I would own a Darter Pro right now, if not for the numpad.

elpete, 2020-04-10:
The new Lemur Pro looks like a good option. :-)

briggs1978, 2020-04-10:
> The new Lemur Pro looks like a good option. :-)

I'm genuinely tempted, but the barrel charger kills it for me.
https://github.com/system76/laptop-suggestions/issues/9#issuecomment-612034771

leviport, 2020-04-10:
@briggs1978 It charges over USB-C as well