system76/laptop-suggestions #23: AMD Radeon Navi GPU

pcewing, 2019-11-22:
### Why/User Benefit/User Problem
The drivers for the new Navi GPUs are 100% open source and included directly in the Linux kernel as of 5.3. For users who prefer not to run binary blobs this is a huge win.

It’s also great for anyone who wants to play games but also run any of the wlroots based Wayland compositors such as Sway. Currently, wlroots only works with the open source nouveau drivers but those perform terribly in comparison to the proprietary drivers.

### Description of the feature
Sell one or more versions of the laptop that come with a Navi series GPU such as the RX 5500M instead of an Nvidia GPU.

dennis1248, 2019-11-23:
I am in support of this. AMD muxless hardware is significantly easier to work with from my experience then that from Nvidia. The AMD stuff just works, the Nvidia hardware however is prone to various issues which the user will have to work around via kernel parameters. Even if you do get it to work Nvidia is prone to other issues with Wayland and what not.

fanhir, 2020-02-07:
I am in support of this. Anything with an Nvidia GPU is not an acceptable option.