system76/laptop-suggestions #24: Fingerprint reader or facial recognition authentication method

thenexus7, 2019-11-22:
### Why/User Benefit/User Problem
Password authentication on laptop can be a pain for end-users, especially when you're forced to login many times per day. Then, fingerprint readers support on Linux is always a bit problematic, mostly on hardware support side since most manufacturers uses and rely on proprietary devices (Yes I'm talking about you Synaptics). 
### Description of the feature
System76 could be a step ahead their competitors by being the first to offer a open firmware based authentication hardware, this can be a fingerprint reader of a facial recognition camera (have a look at Boltgolt's Howdy here for this: https://github.com/boltgolt/howdy)

When you have a Linux supported fingerprint reader (Lenovo T450s) and use Pop!_OS it's a real charm! It's well supported through Gnome and make login and installing apps/updates a super seamless experience!

miharkula, 2019-11-22:
I second the fingerprint reader

fleskesvor, 2019-11-22:
Fingerprint reader would be great. My work PC is a ThinkPad P51, which does technically have one, but alas, it is completely useless on Linux.

bradleeedmondson, 2019-11-23:
Consumer grade fingerprint readers and facial recognition software tend to be pretty insecure in my opinion. I would rather System76 not add those kinds of biometric features.

Additional support for hardware tokens, though (perhaps in UEFI/pre-boot) would be interesting though. Perhaps CAC or USB FIDO.