system76/laptop-suggestions #25: Thinkpad style trackpoint

rik-shaw, 2019-11-22:
### Description of the feature
A good trackpad is valuable, but many prefer the thinkpad-style "trackpoint" ("joystick" between the ghb keys plus physical buttons below the spacebar). Don't follow the apple way of only having a "magic trackpad". Lenovo briefly followed the "no real buttons" path on their t#40 series, but corrected their ways starting again with the t#50 series and onward.

mtkunkel, 2019-11-22:
Yasss!!!   No idea what is patented / tied up from an IP standpoint.    It is much preferred to avoid using the mouse at all (keyboard nav) but when necessary, it's the trackoint and **3** buttons under the space-bar.   Middle-click is best click!

Kokokokoka, 2019-11-26:
Also it would be great if it had thinklight/backlit keyboard too.

styr0, 2019-12-05:
To my knowledge, no other System76 laptop has this yet. While I prefer to use keyboard only and no pointing device, I do not always have that luxury when I need to access remote Windows boxes/VMs, etc., to use proprietary GUI software that actively thwarts accessibility.

Thus, lack of a pointing stick is my main dealbreaker. If System76 came out with ThinkPad-style TrackPoint on the new laptops, or any of most existing models, that would put such laptops at the top of my list for new purchases. I would even settle for another type of pointing stick if necessary.

This should at least be an option (possibly along with a miniature trackball!). Bonus points if there is a completely trackpad-less option :)

mntwinsfan, 2019-12-27:
I would also love to see the trackpoint added (just like Lenovo).  This is the biggest thing holding me back from buying System76.  I can't use the trackpad and won't buy anything without.

fanhir, 2020-02-07:
I would also very seriously consider a System76 laptop if a trackpoint is available.

zearen, 2020-02-12:
I have a Wild Dog and a Galago Pro, and am very happy with them.  I've been on the fence about replacing my old Galago Pro with one that supports USB charging, but a System76 laptop with a track point would push me over the edge and I'd buy it immediately, and maybe another for my girlfriend, too ^.^

emanuelserpa, 2020-07-09:
system76 is losing money for not having a laptop with trackpoint.

HiPERnx, 2020-07-09:
> system76 is losing money for not having a laptop with trackpoint.
> 
> Many thinkpad enthusiasts would love that.

I'm currently looking to replace my old Thinkpad and want to like the Syste76 laptops, but trackpoint is a requirement as I'm a long time Thinkpad user. I hope to see a trackpoint System76 laptop in the future.

achkasov, 2020-11-12:
> Yasss!!!   No idea what is patented / tied up from an IP standpoint.    It is much preferred to avoid using the mouse at all (keyboard nav) but when necessary, it's the trackpoint and **3** buttons under the space-bar.   Middle-click is best click!

The patent has expired in 2017: [Google Patents: Trackpoint device - Lenovo PC International Ltd](https://patents.google.com/patent/US6115030A/en)

Jacharcus, 2021-01-11:
I think this should be a consideration especially now that System76 will build their laptops in-house instead of basing them on Clevo designs. I would certainly buy one in a heartbeat!

nrvale0, 2021-02-13:
I'm due for a refresh and am here via search "Which System76 laptop has a trackpoint?"

Add me to the pile.

articlecat, 2021-04-14:
Yes, please include a pointing stick on the keyboard, along with three buttons (left, middle, and right) immediately under the keyboard. A keyboard equipped with a pointing stick is the most efficient tool for handling tasks that combine mouse and keyboard input.

morganchristiansson, 2021-04-14:
[image]

deepfire, 2021-04-18:
Any official response at all? : -)

leviport, 2021-04-19:
> Any official response at all? : -)

There won't be official responses to these suggestions, but they are all seen and appreciated.

LordBin, 2021-05-02:
A trackpoint would be awesome, even better if they just removed the trackpad as an option. Imagine all the space saved by not having giant trackpad in the way!

Yethal, 2021-06-07:
Literally the only reason I haven't switched over from a ThinkPad yet.

asitha-sena, 2021-07-01:
Lack of a track pointer is the only reason I'm not switching from a ThinkPad. Hope it gets added eventually.

eckelj, 2022-07-07:
I need a trackpoint, it's THE one condition that I have for laptops. have been using them since my first TP and besides... the nice stuff you have, the only aspect that prevents me from placing the order.

jconnolly837, 2023-01-26:
I can only really be happy with a touchpad for casual use. When it comes to productivity tasks like video/image/audio editing, word processing, and writing code, I can't imagine not having a trackpoint. God knows how much more life I can squeeze out of my beloved Dell D620. It's been an awesome nerds laptop and daily driver for such a long time. But I'm starting to desire a laptop that I can just buy, use, and not worry about. And it must respect the user which rules out Windows, Mac, and ChromeOS laptops. System76 is perfect because they make both the hardware and software... well, would be perfect if there was an offering with a trackpoint.

jebotz, 2023-04-05:
Upvote from me.  It's the main reason I still stick to Thinkpads.  Although Apple's trackpad comes cose to converting me, I still prefer to keep my hands on the home keys, and in any case I'm a Linux guy and MacOS leaves me meh.

Other key issues for me are keyboard quality (Thinkpads' has been going downhill, but even so is still better than most) and easy availability of replacement parts (keyboard and battery mostly).

Happy-Dude, 2023-06-15:
Seeing the following tweets

https://twitter.com/carlrichell/status/1662112169360773121
https://twitter.com/jeremy_soller/status/1662552408906694657

makes me think the System76 team is taking a trackpoint/pointing-stick keyboard seriously for `Virgo`.

I have an idea for the color: "stardust"
When I imagine it, its some mix of grey-gold-brown. The System76 [brand](https://katehazen.myportfolio.com/system76-rebranding-process)/logo already uses Pantone 137 (#FFA300), which seems perfect and fits with the established brand.

mkelly1993, 2023-07-08:
I'm in the market for a new laptop and very much want go System76, but the lack of a TrackPoint prevents me from making the jump from Lenovo.

While still hopeful, I'm dubious we (TrackPoint aficionados) are a large enough market to justify the cost of engineering a new laptop design (and possibly covering licensing fees or whatever--not sure what the patent/IP situation is with the TrackPoint tech currently). I wait with bated breath...

benkay86, 2023-08-04:
+1 for trackpoint, recently tried a laptop with trackpad only and hated it.  All of my future laptops will have trackpoints.

oldlaptop, 2023-10-29:
I have no clue what the "Virgo" is or is meant to be, beyond this photo: https://fosstodon.org/@carlrichell/110571781656976874

If it ships with that (assuming reasonably good TrackPoint hardware behind it), you won't be able to take my money fast enough.

wdbm, 2024-04-21:
Three buttons and a pointing stick.

bryankaplan, 2025-06-16:
A hundred upvotes on this suggestion. The lack of a trackpoint has always been the only factor preventing me from considering the purchase of a System76 laptop.