system76/laptop-suggestions #26: MBP level battery life

HankG, 2019-11-22:
### Why/User Benefit/User Problem
Battery life is probably the biggest glaring problem I see in System76 laptops. 4 hrs or less just doesn't cut it for travelers and people who have to jump room to room for meetings etc. throughout the day so stay unplugged for hours at a time.

### Description of the feature
For me a 13" ultrabook for travelers with a battery life comparable to an MBP or XPS13 is my ideal next laptop that I'd buy from System76. When it comes time to get a new production Linux laptop the only one in the running right now is XPS13 because of it. Can I work cross country on a flight without a charger without the battery dying doing writing and/or light development? 13" MBP gets me 10+ hours of light writing and 6+ hours of development. XPS13 is comparable from what I'm reading.

15" model with the same battery life would be useful for office use where disconnecting from power for meetings but needing more screen real estate and horsepower are key.

rik-shaw, 2019-11-23:
I agree but would like to see Thinkpad level battery life of 14+ hours :-)

ParaplegicRacehorse, 2019-11-23:
I agree. There's just no way I'm spending +USD1000 on a mobile device with less than 8hrs battery life.