system76/laptop-suggestions #29: Battery Live > everything else

q5sys, 2019-11-22:
Please don't do the same thing every laptop maker is doing by chasing the 'Thin' trend.  Some of us actually use our laptops all day long and need battery life more than we need to have a laptop that's 2 to 3mm thinner.  There's times Im working in the field where I may only see an available outlet once a day and when I do everyone is fighting over it.   
Battery life is a feature not an afterthought.  
I know the FAA has limits on wattage per battery, so go the lenovo route and have an internal battery which cant be removed and an external battery that can.

TheLinuxNinja, 2019-11-22:
Loving my system76 Gazelle... wishing for more cowbell, I mean, battery. I can go nearly 2hrs, but an external battery that I could attach and get me through the day would be a life-saver. I hate going into a conference room for a 2hr meeting, and I'm the guy who has to sit by the outlet in the back of the room.

mtkunkel, 2019-11-22:
Collecting T480 models just for this reason...   Latest hardware with hot-swap battery.  Availability of 72Wh packs for all  day travel hacking.

q5sys, 2019-11-22:
Right now I've got an x250 and I carry 2 to 3 extra batteries in my bag.  I'd like to find some more OEM 72Wh batteries, but all I can find are Chinese knockoffs.

Dmole, 2019-11-28:
OLED with a dark theme (maybe even on top of e-ink) would be a big battery life extender for those hot swap-able 100Wh batteries.

eddyp, 2019-11-29:
The Apple PowerBook G4 had the possibility of hot swapping the battery and since the laptop would go into sleep on low battery levels or when you closed the lid, you would not lose any data, but you could continue work for yet another 2 hours if you had an extra battery.

Today I think 4-6 hours on an extra battery should be doable.

aaronfranke, 2020-07-02:
What is the current situation with System76 laptop battery life? Out of their current laptops, what battery life does each have?

Dmole, 2020-07-02:
> What is the current situation with System76 laptop battery life? Out of their current laptops, what battery life does each have?

"Up to 14 hours of battery life per charge"
https://system76.com/laptops/lemur