system76/laptop-suggestions #30: No Biometric Authentication

DontUseGithub, 2019-11-22:


Gabgab2003, 2019-11-23:
Could you elaborate?

ericmann, 2019-11-23:
Effectively, done support either fingerprint or face scans to unlock the machine. I tend to agree, but would argue either (or both) feature still has value _after_ the system is unlocked...

bradleeedmondson, 2019-11-23:
@ericmann Meaning, System76 should _not_ support biometric auth hardware, due to privacy and accuracy concerns? Or the _problem_ is that there is no biometric auth, and they should add it? I'm unclear on the verbiage as it currently stands.

I tend to fall along the lines of the former, esp. w/r/t consumer grade fingerprint and face recognition, but support for military grade hardware tokens like CACs (although not perfect) would be more interesting.

ericmann, 2019-11-23:
My Oryx has a non-functional fingerprint reader. I either want the hardware to work or not be there. And while I'd love for it to be used by apps for secondary identity verification I do NOT want it to unlock my machine. Same with face unlock - secondary identity verification sure, but not system unlocking.

Dmole, 2019-11-28:
A fingerprint reader would be a nice **option** if it was reliable (some are really bad, others almost perfect)