system76/laptop-suggestions #38: eGPU support

LeviMWilliamson, 2019-11-23:
### Why/User Benefit/User Problem

Discrete graphics can drain battery life quickly, be make a laptop heavy, and present massive heating and power-consumption problems. They also create a lot of driver problems because if you have both integrated and discrete graphics, there needs to be some switching behaviour which decides which internal device is operating at a given time.

Providing support for external graphics extends the lifespan of a laptop by making it more modular. It also lets you forgo discrete graphics without severe loss in functionality.

### Description of the feature

Have a Thunderbolt 3 port which works with current marketed eGPUs.

Bonus: Enable external devices to send display output to the laptop display via HDMI or thunderbolt.

bradleeedmondson, 2019-11-23:
Laptops with a current Thunderbolt 3 port do support eGPUs. I have a Radeon 590 attached to my galp3 (2018 model).

(I don't believe it's fully documented, but it does appear to work -- that is, it works, and I /think/ at full 40gbps pipe.)

LeviMWilliamson, 2019-11-23:
You're right. There's clarifications on the Thunderbolt website confirming this (here)[https://thunderbolttechnology.net/blog/the-difference-between-egfx-and-egpu].

I think this part is still relevant: 
> Intel’s Thunderbolt 3 certification now includes verification eGFX functionality for devices and host systems. We do not validate individual components, for example the eGPU. The full validation is the responsibility of the eGFX device and PC manufacturers.