system76/laptop-suggestions #39: A model aimed at virtualization and compilation (All CPU and memory, no secondary GPU)

dennis1248, 2019-11-23:
Long story short; powerful 6 or 8 core CPU, 16GB of RAM upgradable to 32GB minimum and NO secondary muxless switchable graphics. 

This is a machine which is aimed at people who need CPU and memory above else. I have noticed that everything available on the market which runs a CPU powerful enough to do such tasks also has a secondary graphics chip which does nothing but inflate the price and increase heat production when it does engage.

bwijsmuller, 2019-11-23:
64gb (or more even) would be a unique selling point! And yes I would benefit from more ram.

FiretronP75, 2019-11-28:
I agree. GPU is useless to many of us. There should always be a no-GPU option.