system76/laptop-suggestions #47: ARM

ParaplegicRacehorse, 2019-11-23:
If Microsoft (!) and Apple (!) are pushing it, then ARM is the future. Go all-in for ARM.

(But, in the next 3-8 years, keep an eye on RISC-V!)

mattwhiteside, 2019-11-26:
+1 to RISC-V.

Dmole, 2019-11-28:
AFAIK there is only 1 ARM laptop with >=8GB RAM.
Drivers for ARM are historically slow to emerge so that may be another issue.
But yes I do want that 32 core non-speculative RISC-V open driver Laptop,
or POWER9 on 7nm lithography because it's already developed, open, available, and endorsed by the FSF.

bryceadamprescott, 2019-11-28:
Looks like there is currently attempt going for a PowerPC Linux laptop: https://www.powerpc-notebook.org/en/ (not sure if it adheres to POWER9 or not). Currently, it seems like most non-x86 architectures are more open on the software end with RISC-V, of course, being the most open. ARM would probably be the fastest performing, best supported, and most affordable non-x86 option. Don't get me wrong, I love the idea of RISC-V.

phma, 2019-11-29:
I'd prefer POWER9 to ARM. It has four threads per core and a 128-bit long double type. The ARM has no float longer than 64 bits AFAIK (I've run code on a Raspberry Pi and it had nothing longer than 64).

I'm planning to get a Talos II running Adélie for big-endian portability testing.

ParaplegicRacehorse, 2019-11-30:
> AFAIK there is only 1 ARM laptop with >=8GB RAM.

At the moment, yes. Apple already announced that their Macbook Air line is switching to ARM in Q1 2020, with possibly the other Macbooks soon to follow. And MS has already released (September) an ARM-powered Surface. HP, Dell, Lenovo, Acer are already in also; and all planning to release bigger-badder models.

> Drivers for ARM are historically slow to emerge so that may be another issue.

I think Google's announced plan to push their Android kernel customizations up to Mainline will make a huge difference. Also, there's lots of ARM coming in 5.4 and 5.5. Probably the biggest hurdle will be GPU drivers, though; especially the proprietary blobs (for those of us who want them.)

I'm also digging on the Power9 thing, but the amount of user-space software already available in ARM repositories makes it more attractive to me. YMMV.

Dmole, 2019-12-01:
missing VPU drivers is more of an issue than gpu.

> ... Power9 ... user-space software...

seems to not be an issue;
https://www.phoronix.com/scan.php?page=article&item=power9-linux-distros&num=1