system76/laptop-suggestions #50: Aluminum body/frame without sacrificing utility

Gman0064, 2019-11-24:
### Why/User Benefit/User Problem
Most people who invest in a brand new performance laptop (like myself) are dropping a significant amount of money for a machine that they expect to be very durable while also looking professional. An aluminum body similar to that of a 2012 MacBook Pro or Razer Blade, where the majority of the body is a solid piece of aluminum with a faceplate at the bottom for hardware access, would make for an amazing Linux alternative to the aforementioned options. The professional and eye-catching design of the aluminum body combined with a tailored Linux distribution that works 100% with the hardware out of the box would make for an absolutely killer machine. 

### Description of the feature
Have the body of the machine be built out of some metal or aluminum, while still allowing the user to be able to have access to the innards of the machine. An aluminum top half of the machine with a faceplate on the bottom to allow for easy access to the hardware would be very ideal. Of course, it would also include ventilation at the bottom to prevent overheating.

pandiloko, 2019-11-25:
I still don't fully understand this trend of phones and laptops having to be made of aluminum and crystal instead of (good quality) plastic. I have well functioning Thinkpads made of plastic which predate any old alu MacBooks I know off and I'm sure I am not alone. A good quality well built plastic body is also more flexible and resistant to falls and scratches. My 2 cent.

eduncan911, 2019-11-28:
> I still don't fully understand this trend of phones and laptops having to be made of aluminum and crystal instead of (good quality) plastic. I have well functioning Thinkpads made of plastic which [...]

First, they don't make thick hardshell plastic like those old those old Thinkpads any longer.  My last 3 thinkpads ranging up to last November's P1 beast is think flemsy bendable plastic.  Even the System76 laptops I've seen in person follow this thin plastic twistable trend.  Feels cheap.

Where as my 2012, 2014, 2015, 2016, and 2018 model MacBook Pros (yeah, I've had a lot for work) are all extremely solid - no flex.

Dmole, 2019-11-28:
It would suck to close the screen to fast and have the plastic flex enough to snap the screen... unless the screen were flexible OLED then it may not matter so much, until someone sits on it.... Zack should start testing laptops on JerryRigEverything.

pandiloko, 2019-12-04:
I must admit the most recent Thinkpad I have is a 2012 T430s from which I am typing right now, so no experience with more modern, thinner versions like the new Carbons and the like. A friend of mine has a 2018 Dell XPS though. Also made of plastic IIRC, but it didn't feel cheap to me at all. 

I guess maybe it is a taste thing but I often see macbooks from colleagues with scratches and slightly bent in the corners. I mean, it seems to me there is no clear advantage in using such "premium" material. If it is cheaper with plastic, I would rather use that and invest more in raw performance, better screen, etc.

Gman0064, 2019-12-04:
Currently using a Dell XPS 9570, and even though I like the rubbery-smooth plastic in the inside of the machine, I would much rather have an all-aluminum shell as opposed to the aluminum outside/plastic inside approach. Again, this is a matter of personal preference, but I think if System76 was to develop a product that matches some of the elements of the MacBooks build quality while maintaining options for power users (hardware accessibility, no TPM nonsense, eGPU support), I think this would introduce more people to the market of Linux hardware that would otherwise not be interested in, which I think should be something to consider. 

In my opinion, there isn't really any hardware on the market that comes pre-installed with a popular Linux distribution that consumers could easily adopt right now (Dell XPS can be ordered with Linux, but you have to specify it). Sure, you can raise the argument of PineBooks coming preinstalled with Linux and Chromebooks being widespread Linux machines, but those machines are purpose-built: Pinebook is an ARM laptop designed for power efficiency and light work, while the Chromebook is a fully customized machine that on a standard, surface-level doesn't share any of the operability features that Linux does and does everything in a "Google-designed" fashion. The Chromebook is designed to be as approachable as possible, and I would argue that this is the reason it is so successful. Of course, System 76 isn't going to make a Chromebook (hopefully), this is supposed to be a fully-fledged machine for power users, which is why I brought up the design of a MacBook.

Apple's Macbooks are popular for a reason. Disregarding the obvious of being produced by Apple, for the majority of people they are aesthetically pleasing and durable machines, and people buy them for that reason. Personally, I would definitely buy a "Linux Macbook Pro" if such a thing existed, with the physical design and feel of a MacBook but with the operability and customization of a Linux machine. I think if System76 could produce a polished product that can service both the power-users and the average developer/ease of access crowd, whilst maintaining the accessibility that makes these other products so successful, System76 would have a goldmine of a product.

FiretronP75, 2019-12-04:
All I care about is thin and light. I don't care which material achieves that.

As for attracting non-Linux users. That is a tough sell no matter what you do, because people tend to get unreasonably attached to their software, logo, and even OS. I believe the best approach is what System76 is already doing: advertising privacy and security.

I'm only considering Purism and System76 for my future computers, because they are the only 2 manufacturer's I'm aware of that are fighting against both hardware and software back-doors.

Public awareness of "everything spies on you" is growing faster than public awareness of "open-source software is just as good or better". And "hey this looks just as classy" isn't a way to convert people over. Matching design looks could remove a barrier for some, but not provide a reason to switch. And even and exact copy of a Macbook, would be frowned upon as a knockoff. Because people mostly like Apple just because of the feeling of the brand, not for anything practical. So getting something "just like" Apple would just make such people feel like they bought a knock-off.

anilsg, 2019-12-26:
I also want to show support for any combination of materials that System76 expertise finds best to provide a strong and cosmetically appealing build. I've got a 2017 X1 Carbon that's been very durable and pleasant. I like the Dell XPS. If System 76 has something unique that has similar integrity and durability I expect a lot of people would find that cool.