system76/laptop-suggestions #54: 14 inch / with WQHD and up display options

jwuensche, 2019-11-26:
### Why/User Benefit/User Problem
The system I have in mind would be suited to developers who use their notebook for a long timespan daily and prefer to have light system they can take with them and use while travelling e.g. by train.

The main benefit would be portability while still retaining performance.

### Description of the feature
#### Display
I always enjoy working on higher resolution and would love to have a display in my laptop that has at least 1440p. While this would be optimal (less scaling issues in desktop environments) higher resolution like 4K would be also okay.

#### Size
Most of the time I carry my Laptop with me, so I prefer to have smaller display size which I can still work with all day, so around 14 inch are a good fit.

#### CPU
Though I prefer AMD as the vendor (less security bugs and hardware issues), Intel would also be okay. The current generation of 4 core CPUs with low voltage seems to perform quite good.

#### GPU
A definite no for NVIDIA GPUs, only encountered troubles with their drivers, which made me switch my system recently.
With AMD GPUs my experiences are that they work much better, but because of power consumption I would just leave them out.

#### RAM
Unsoldered, with at least 2 Slots. The maximum capacity is at best 64 GB.

#### Ports
USB-C ports for charging and transmission would be nice, with thunderbolt3 support (if shipped with intel). If other dedicated ports would be added this is also fine, e.g. for Video (miniDP) or Audio (3.5mm jack).

#### Battery
The battery should be aimed to hold at least 6-8 hours with powersaving on.

#### Keyboard
Perfect would be keyboards with higher key pressure required such as the Thinkpad ones.

mattwhiteside, 2019-12-10:
I think the 14" screen size is ideal.

algobolson, 2019-12-26:
upvoted because I'm a happy current owner of a HiDPI Galago Pro (3200x1600 iirc) and I'm sad to see that isn't a current option. 14" is the right size of machine and I love all those pixels.

jonpolak, 2019-12-27:
> I think the 14" screen size is ideal.

Not to mention, free of PWM of course