system76/laptop-suggestions #56: Working hibernation and (deep) suspend (last one, I swear)

bryceadamprescott, 2019-11-26:
Please make sure the hardware chosen can successfully hibernate and suspend (deep not s2idle) without losing WiFi, etc. when waking. Also, make changes to Pop that automatically setup hibernation properly. It's crazy that getting hibernation to work requires some not insignificant tweaking on any distro.

eduncan911, 2019-11-28:
I actually opened issue https://github.com/pop-os/distinst/issues/171 about this in their setup utility. I think I am overdue to write a huge blog post about how to do it with Pop_OS.

bryceadamprescott, 2019-11-28:
Good on you for informing System76 of that shortcoming. I can get hibernate to work using manual partitioning on almost any distro. The problem with most distros is that the automatic partitioning creates too small of swap partition.

eduncan911, 2019-11-28:
The biggest change is to get them to move the swap to LVM instead. It's not bad the way they have the small swap partition which is actually secure because every boot gets a new random key for the swap.

However, that prevents hibernation.

I posted a solution to this, as well for machines with small HDDs which seems to be there only concern.

If someone can point me to the code in that repo, I'll modify it for the propose solution.