system76/laptop-suggestions #57: Remove keypad, re-center mousepad with screen

stmurphy65, 2019-11-26:
### Why/User Benefit/User Problem
Off centered mousepad causes health problems in wrists and forearms.  You can't properly seat the laptop centered on your lap (where the screen is centered) and comfortably operate the keyboard and mousepad due to the addition of the keypad.  

### Description of the feature
The addition of the keypad is an unwanted item by most people.  While there are few who do like it, lefties would argue it's useless to them being on the right.  Off center mousepad and keyboard causes medical issues (wrists and forearms) when centering the laptop on ones lap.  The ideal placement of a laptop is to have a centered screen with ones head.  Having to place it to the right to center the keyboard and mousepad causes neck issues from always looking to the right and the laptop does not stay seated properly due to the lack of surface area upon the left leg.  The laptop keeps falling down - try it on a shaky/bouncy bus sometime - it's hell.  

The solution is to get rid of the numeric keypad (make it external via USB-C and "optional" for purchase, making everyone, especially lefties, happy) and then re-center the keyboard and mouse pad with the screen. Fill in the extra space on each side of the keyboard with up-facing speakers, which is a better location for them anyway.  See a MacBook Pro for how it should be done the right way. (Apple laughs at the current laptop designs) (I'd buy a MacBook Pro but I'm not a fan of apple)

mattwhiteside, 2019-11-27:
I agree with this fully.  I'd maybe take it even further, and question the value of the right-most column of keys on the Galago Pro -- PageUp, PageDown, etc.

Digitalman42, 2019-11-27:
> I'd maybe take it even further, and question the 
> value of the right-most column of keys on the Galago Pro

I totally agree. I had intended to mention it  but forgot. Thank you for including it.  Removing those would then allow for complete centering of the keyboard and mousepad with the screen.

dtbeltramo, 2019-11-28:
The never-used 10-key pad also makes the laptop wider than necessary, so it impacts portability.

jpritikin, 2019-11-28:
I'm had a laptop with the mousepad centered relative to the laptop chassis. If the keyboard is centered too then it's okay. However, if the mousepad is not centered relative to the spacebar then my wrists tend to rest on the mousepad and it can cause random mouse movements. I know there is special software logic to suppress mouse activity during typing, but it doesn't work perfectly and hurts perceived responsiveness. All I'm saying is that the mousepad must be centered relative to the keyboard. Another option is to provide a [pointing stick](https://en.wikipedia.org/wiki/Pointing_stick). If there is a pointing stick then the mouse pad can be disabled if it causes too much random movement by accident. But better have a good design for the mouse pad so it works without picking up accidental movements.

eddyp, 2019-11-29:
Yes, please! Tenkeyless laptops should be a thing even for >15" screens.

BTW, I don't agree with dropping the pgup,pgdn,home,end,ins,del keys since the alternative is fn+arrow keys or not having them at all, which is a real pain for programmers like myself.

Maybe an alternative is to have F1-12 on one side and the vertical row on the other, but it is still asymmetric (12 F keys vs. 6 on the column) and it would need some 6 other keys for visual balance or a gap. Not sure what lefties would think about this.

FiretronP75, 2019-12-02:
I don't need the number keys. But I do need the Home, End, Delete keys.

vcahkjsa, 2019-12-05:
I support @FiretronP75 idea plus desktop processors and rtx graphic cards with a smaller case. I will buy it asap

sdwolfz, 2023-04-05:
I have to disagree with this one. Numerical keypad is extremely useful for people that have to do financial work or numerical data entry.

In fact there is no point on forcing people to have one or the other. Look at framework's design for the 16 inch. You buy them separately and assemble them where/when needed, and you can centre the touchpad yourself.

Since they've done it then there's no reason you can't do it to. Create a 2 art keyboard and let end users buy the parts they need.

pmontrasio, 2023-04-05:
I'm firmly in the no number pad field: never used it, but I write software. Friends more in management roles say that they use the number pad all day long.

Solution: do what Framework is doing with their next laptop. They'll let people replace the keyboard with one without numberpad. Then you have to let them move the touchpad below to where the space bar gets to be. I think they have a video about that.