system76/laptop-suggestions #60: 15 inches with WQHD or better

bengsparks, 2019-11-28:
### Why/User Benefit/User Problem

The ideal system for me would be a portable laptop with a GPU that can be used for business and work, even on the go, and light gaming in my free time.

### Description of the feature
## Display

Working with higher resolution decreases eye strain while still displaying a reasonable amount of information on the screen.

## Size

The laptop would have to fit in a rucksack, so a ceiling of 15 or 17 inches would be appropriate.

## GPU

Please don't use NVIDIA.

##  Ports

USB-C ports for charging and transmission would be nice, with Thunderbolt 3 support (if shipped with Intel)

jpritikin, 2019-11-28:
I agree that resolution higher than 1080p is nice (WQHD is 2560 x 1440). Ubuntu support for HiDPI currently has some shortcomings, but these are getting solved gradually.

Dmole, 2019-11-28:
Would eGPU support fit your needs?

bengsparks, 2019-11-29:
eGPU would conflict with the wish of portability imho, unless you mean keeping the eGPU at home

loligans, 2019-12-09:
I would like a 15-inch with no dedicated GPU but have eGPU support for when I dock at home.

anilsg, 2019-12-26:
I've +1 this one because I'm looking for a dev workstation that can do some light gaming as well, like the recent X1 Extreme. I'm usually happy with 14" and want to hold down the size and weight for true portability. I don't know about gaming hardware, would appreciate knowing why "not Nvidia". Perhaps I shouldn't have +1 because I believe high resolution is going to hit the battery life and on small screens I'm not sure the higher res is useful. Also since can't expect high power gaming machine it's not going to be able to drive WQHD! I think this is aiming too high and can't really be a serious request for a dual purpose business / gaming since evidently it has to be mobile and the gaming can only be expected to be "light". I don't see a big advantage in WQHD gaming either and see that as high end gaming anyway. I think there are contradictory goals in this one but I definitely support adding a graphics card. I wish the bug title was different.