system76/laptop-suggestions #61: RedoxOS

Dmole, 2019-11-28:
### Why/User Benefit/User Problem
Security!!!

### Description of the feature
Kill 80% of bugs right out of the gate by using a modern language like rust

https://www.redox-os.org/news/focusing-on-rustc/

FiretronP75, 2019-11-28:
It looks very promising! But, I don't think it is mature enough for System76 to mainstream it yet.

Dmole, 2019-11-28:
@FiretronP75 true, but it would be nice if they helped the initiative along. Maybe $1 from each purchase or some developer time if they are already working on drivers. Maybe if they can get to the point of running Linux in a redox VM (QubesOS style).

ilyaigpetrov, 2019-12-01:
I would like to opt out from this and save my $1 for something else.

Dmole, 2019-12-01:
> save

I'm sure they could write it off so as not too cost the customer anything.

Kokokokoka, 2019-12-01:
Writing X in rust doesn't make X secure. It's a myth/False adv/you name it.

Dmole, 2019-12-01:
> ...rust doesn't make X secure...

I claimed 80% more secure (not 100%) the claim is from historical bugs that could not have happened if modern language saifty were observed. Needlessly coding with out saifty is like clif climbing without ropes.

Obligatory xkcd-s;
https://xkcd.com/378/
https://xkcd.com/2030/

Kokokokoka, 2019-12-01:
So more buzzwords without a proper claim. nice! 
Can you _formally_ prove your statement (I guess not)?
And also, how is rust different from f.e. safeD?

Kokokokoka, 2019-12-01:
Please, be so kind and open page 16 in pdf that you've provided.

Dmole, 2019-12-05:
> Can you _formally_ prove your statement (I guess not)?

100% (N=10);
```
#!/bin/bash

#
# cve_scrape.sh
#

#https://nvd.nist.gov/vuln/data-feeds

curl -s https://nvd.nist.gov/feeds/json/cve/1.1/nvdcve-1.1-modified.json.gz \
	| gzip -d \
	| jq '.CVE_Items[] 
		| select(.impact.baseMetricV2.severity == "HIGH") 
		| select(.cve.description.description_data[].value 
			| contains("Linux kernel")) 
		| { severity: .impact.baseMetricV2.severity, 
			id: .cve.CVE_data_meta.ID, 
			description: .cve.description.description_data[].value } '

echo "On 2019-12-05 there were 102 Linux kernel high impact security vulnerabilities
Here is a list of the last 10 Linux CVEs and wheather Rust Could have Prevented(RCP) them;
ID,             RCP, Type
CVE-2019-9112,  yes, overflow
CVE-2019-9111,  yes, overflow
CVE-2019-9003,  yes, use-after-free
CVE-2019-8980,  yes, memory leak 
CVE-2019-8912,  yes, use-after-free
CVE-2019-3896,  yes, double-free
CVE-2019-19083, yes, Memory leaks
CVE-2019-19082, yes, Memory leaks
CVE-2019-19081, yes, memory leak
CVE-2019-19080, yes, memory leaks"
```
@Kokokokoka Your turn.

Kokokokoka, 2019-12-05:
>100% (N=10);
Wow, judging from a sample from 10 elements, such a nice statistics you've got here.
But let's not talk about this sample and imagine for a moment that I accept this (even though you CAN'T make a statistics from a sample of 10 elems)
the H_{0} was = "Kill 80% of bugs right out of the gate by using a modern language like rust". Bugs != CVE's there are many other bugs. And design bugs are bugs too. https://ece.uwaterloo.ca/~lintan/publications/bugchar-emse14.pdf This pdf had a nice preview of a good statistics. And you can clearly see that mem bugs weren't the most common ones, heck, they're one of the least common. they've got a big impact though (and I agree with that). My main point is no language will help you designing good apps, it could prevent _some_ (not even close to 80%) of the bugs. 
And my answer is: qmail, libsodium.

Dmole, 2019-12-05:
The first word I wrote in this thread was "security".
I clearly showed how 100% (N=10) of the most recent Linux security bugs could have been prevented with rust.

You made a point that no language (English included) can save some idiots (me, you, them) from themselves, with which I have to agree but to clarify that is not the focus of this thread. I'm not including incomplete features in my use of the term "bugs" in this security-focused thread.

Kokokokoka, 2019-12-06:
security != CVEs, just wow.
Do you understand the word DiD? there are other type of bugs which Rust will never prevent and Ada or idris (for example) will. Let's write a small theorem proving lang and a microkernel! also, thy most popular os is MINIX (intel me)
I haven't seen any of the memory CVEs in it. There were problems, but not with MINIX, so, maybe this is more a design problem and how well one reads code and writes? this is a great project too https://bearssl.org/index.html
>I clearly showed how 100% (N=10) of the most recent Linux security bugs could have been prevented with rust.

And I said that sample of 10 proves nothing as it is statistically incorrect.
And you haven't read other points.
If we had kernel written i idris and ada there would have been even less problem. Time for functional languages to shine!

Dmole, 2019-12-06:
@Kokokokoka you're going to need compelling data (not just talking about tangents) to convince me the safety of modern languages would not benefit kernel/os security.

> security != CVEs

CVEs are the best data available for the purposes of analysis. commit history lacks relavent deductions. Yes CVEs are almost useless for informing patch and upgrade decisions, but thats a tangent.

> OpenBSD/OpenSSH

Has a great security record, but it's slow moving and small relative to Linux. I'm sure if I looked there would be a few places a safer language would have helped even this idealistic project, like this;
https://nvd.nist.gov/vuln/detail/CVE-2019-16905

> there are other type of bugs which Rust will never prevent and Ada or idris (for example) will. 

I have never looked into either of those languages but I assume Mozilla had a reason for Rust, and there is no OS based on Ada/Idris. (I'd be interested if you know of any spesifics)

> MINIX
Is tiny / not comparable like l4

> sample of 10

Is better O.R. than your sample of 0. Even if I did all of them for this year you would probably claim it useless due to the cve data source, but as I have countered it's the only data source and it's a good indication even in its imperfect state... If you summarize the critical CVEs for Linux this year and it differ significantly (+-10%) I'll concede my percentage estimation was wrong.

> other points

I'm aware rust is not the solution to all problems however RedoxOS appears to be the most idealistic almost usable OS to invest in at the moment.

> such a reaction from me

Would have been better if it contained some actual data or a preferred alternative. Because currently I am not convinced my estimation was inaccurate.

Kokokokoka, 2019-12-06:
In the previous post there were alternatives
a) langs: safed, idris, ada, ocaml
b) microkernels: minix, mirageos
c) my sample is from the pdf that you've provided. It shows that you're doing misleading hypotheses
d) It was your statement so it is for you to prove it (logic laws).
e) using a sample of 10 and implying that this is a statistically significant is plainly wrong, from stat point of view, from argument point of view too.

>Has a great security record, but it's slow moving and small relative to Linux. I'm sure if I looked there would be a few places a safer language would have helped even this idealistic project, like this;

Slow moving? care to elaborate? Still this is not an argument that i'll take.
I agree, there are mem CVEs in openbsd, but still, it disproves your point of 80%.

>I'm aware rust is not the solution to all problems however RedoxOS appears to be the most idealistic almost usable OS to invest in at the moment.

That's a pure lie. it's only your opinion. redoxos is nothing in comparison to mirage.

>Is tiny / not comparable like l4

this has nothing to do with correctness/etc comparison.

>I have never looked into either of those languages but I assume Mozilla had a reason for Rust, and there is no OS based on Ada/Idris, so also a tangent. 

Appeal to authority, sigh. 
>(I'd be interested if you know of any spesifics)

I provided links in the previous posts. It's a really good start if you'll even skim through those it would take a week (subpages too)
> Is better O.R. than your sample of 0. Even if I did all of them for this year you would probably claim it useless due to the cve data source, but as I have countered it's the only data source and [...]

You disproved yourself using this pdf already, and you ignored this point too.
Do you even read what you send?

+You're blindly skipped my other arguments. So, from this point, I assume that you're trolling.

Dmole, 2019-12-06:
MirageOS does not appear to be usable as the primary OS on a development laptop (the point of this repo/thread)

> ...Code can be developed on a normal OS such as Linux or MacOS X, and then compiled into a fully-standalone, specialized unikernel that runs under a Xen or KVM hypervisor...

I just reviewed 100% of the CVEs for this year for the Linux kernel (that system76 will be shipping) and found 100% of them could have been avoided by rust. 
I'll just ignore any more off topic comments in this thread because in 6 posts you have not offered a realistic alternatives (We agree BSD is a middle ground).