system76/laptop-suggestions #62: Drop Trackpad Buttons, Double the Size of Trackpad!

eduncan911, 2019-11-28:
### Why/User Benefit/User Problem

Very difficult to switch to, and *use*, a System76 machine after using MacBooks for a decade.

This is the single reason my clients have not ordered a few dozen of these machines.  Everyone uses MacBook Pros, mostly with Linux installed on them. No one wants to give up those extra large trackpads.

Especially since Linux hard-codes the "deadzone"  palm rejection crap (the Apple guys have been patching the kernels to get rid of this) of like 20 or 30%.

### Description of the feature

* Drop the trackpad buttons.  
* Double or triple the size of the physical trackpad.

Bonus points for patching the kernel in Pop_OS to remove (or make adjustable) the useless hard-coded deadzones for Palm rejection of such a large trackpad.

Dmole, 2019-11-28:
I try to use keyboard over trackpad, but when forced away from he keyboard I use a mouse to avoid flat finger syndrome. Touch screens are nice and bug but it's also nice to have a fingerprint free screen.

bryceadamprescott, 2019-12-05:
The buttons can still exist; just "hide" them under the trackpad. What I mean is, allow clicking the whole trackpad and then either use location or number of fingers (preferred) to differentiate between left, right, and middle click. For those who like it, make a "tap to click" option.

eduncan911, 2019-12-05:
> allow clicking the whole trackpad and then either use location or number of fingers (preferred) to differentiate between left, right, and middle click.

Most DEs already can do that with libinput drivers.

FiretronP75, 2019-12-05:
I didn't know buttons under trackpad were still a thing LOL, are some laptops still doing that?

eduncan911, 2019-12-05:
> I didn't know buttons under trackpad were still a thing LOL, are some laptops still doing that?

I hope not. My Thinkpad Helix has them under the trackpad, and I can't stand it.

JohnAD, 2019-12-28:
Or better: make it a user option. I specifically choose laptops with buttons.

Thinking big picture: one could simply make a cavity in the top of the case with published screw-mount and usb port locations. Then even third-parties could make different trackpad setups and styles.

eduncan911, 2019-12-29:
> Or better: make it a user option. I specifically choose laptops with buttons.
> 
> Thinking big picture: one could simply make a cavity in the top of the case with published screw-mount and usb port locations. Then even third-parties could make different trackpad setups and styles. Or people could 3d-print holders for other input methods.

Good idea.  However, it would be more involved than that.  You'd need a different palm rest because a large track pad has different dimensions than a track+buttons.

FiretronP75, 2020-01-03:
This is interesting. I decided to try out POP! OS on a laptop that has no trackpad buttons or hard pressing, only soft tapping. It turned out that by default, the OS disables soft tapping the track pad for clicks. So I had to use keyboard navigation to get to the trackpad settings and enable tapping clicks.