system76/laptop-suggestions #64: Light Weight, Low Power, Large Screen

FiretronP75, 2019-11-28:
### Why/User Benefit/User Problem
I need a large screen to do my work. But I don't want too much weight on my legs. Most 17 inch laptops are workstations or gaming oriented. I don't need the extra power. I just need the big screen at a light weight.
### Description of the feature
17 inch or larger screen at FHD or better, with a weight under 3 pounds.
### My ideal build
19 inch FHD screen
Backlit Keyboard
Core i3 processor
16 GB memory
NVMe drive
No extra drive
No graphics card
No speakers
Small or no battery
Weight 3 pounds

Dmole, 2019-11-28:
"no battery" would be bad because UPSs are important. 
The macbook pro is 100Wh and 2.0 kg (4.3 pounds) with all that aluminium and glass, so a plastic version might be able to stay under 3 pounds even with the largest flight friendly battery possible.

Let those gaming people plug in a modular [e]GPU like those laptop base stations.

FiretronP75, 2019-11-28:
I'd be happy with under 5 pounds actually, since that is so rare still for 17 inch.

FiretronP75, 2019-12-05:
I'd be happy with a 17" version of the Darter Pro that has no other changes except for the screen size.