system76/laptop-suggestions #65: drop audio jacks, support bluetooth audio only

jpritikin, 2019-11-28:
### Why/User Benefit/User Problem

I was anxious when I bought a smartphone without audio jacks. However, bluetooth audio really works great these days. If you engineer excellent bluetooth support, I don't really see the need for audio jacks anymore.

### Description of the feature

Remove the physical audio in/out jacks. Use the space for something else.

FiretronP75, 2019-11-28:
But they barely take any space, and many of us insist on wired audio over wireless.

Dmole, 2019-11-28:
Bluetooth audio is really limiting;
+ audiophile things still normally not wireless.
+ wireless is normally expensive.
+ wireless is unreliable (fails at all large gatherings)
+ wireless is disposable (no battary replacement)

The 2 things wireless has going for it is wires not getting caught, and The SNR on low impedance wired connection is often horrific. But let users chose by offering both.

humanamerican, 2019-11-29:
Please do NOT do this! A wide selection of ports when other manufacturers are getting rid of theirs is one of the things I like about System76 computers.

ajflores1604, 2019-12-08:
Please do not follow this suggestion. aux has lots of interesting and fun functionality in the maker space as an interface to send a signal. It's not just about an interface to listen to music.

Thulinma, 2019-12-09:
Also worth mentioning: while bluetooth audio __output__ has acceptable quality these days, duplex operation is an entirely differrent story. I have never encountered a bluetooth headset that did duplex at an acceptable quality level, even under ideal conditions.

dmiller309, 2019-12-27:
I would be for Bluetooth-only if there were a way to easily disconnect my airpods from my phone and connect to my laptop. It takes a lot of manual steps to connect to my airpods when I want to start playing audio:
1. Click the top-right power button.
2. Click Settings to open the gnome settings panel.
3. Click on Bluetooth in the left panel.
4. Wait for the airpods to show up in the list. Click on the airpods.
5. Click the button to play sounds from the Bluetooth device.

Sometimes the pairing fails or the system won't play to the Bluetooth device. Ideally, the system would automatically connect to my airpods when I begin media playback.