system76/laptop-suggestions #66: Large screen: 17" or more

ilyaigpetrov, 2019-11-28:
### Why/User Benefit/User Problem

I almost never use my notebook during travelling/errands. I use it in the same way as I would use a stationary PC for coding/compiling software and running VMs. Coding on a wide screen seems like a good benefit.

### Description of the feature

Provide something like Darter Pro but with 17" screen or more.
I almost don't care about the weight or power.

### Similar requests

[Light Weight, Low Power, Large Screen](https://github.com/system76/laptop-suggestions/issues/64)

Dmole, 2019-11-28:
Why not just use an external screen so you don't need to upgrade the screen at the same time as your laptop?

ilyaigpetrov, 2019-11-28:
Sometimes I move my notebook from building to building. Moving an external screen would be a problem. Moving 17" notebook is not a problem. Also buying one extra screen wouldn't be very cheap.

FiretronP75, 2019-11-28:
No one wants to carry an extra screen around. And you can't just leave an extra screen sitting around every place you frequently visit. And you most certainly don't want to put an external screen on your legs when sitting in a bed or chair with no desk.

ilyaigpetrov, 2019-11-29:
17" would be large enough, I don't think we need larger.

anilsg, 2019-12-26:
I mostly use my laptop with a screen, even if that's an old 17" at one of my locations. If it's being used stationary, office, plugged in, I'll have a monitor, and a chair and desk for health. When I use it mobile I don't want to be lugging 17" around and use has to be limited because of battery life and because looking down to a laptop screen involves some strain. With my 180' hinge on my X1 Carbon I bring a keyboard with me, open the laptop fully, and place it vertically so the screen is lifted higher and I can use that for some time and the 14" is tolerable. I think there's two completely different ideas here. A heavy full screen hardly mobile device for people who don't actually have a monitor and a truly mobile laptop.

cgltower, 2019-12-27:
17" or 18", 3k or 4k, bezel-less, would be awesome. Needs long battery life when on the go. I'd use for work and browsing/streaming whilst mobile, then plug in to eGPU for gaming/3D work.

JohnAD, 2019-12-28:
17" screen. (or, in my dreams, a 19" screen, but I understand the industry dislikes laptops that big.)

I'm using a Serval WS with 17" now and there is no way I'll ever go to a smaller screen. But I'm just one voice, so I understand if your first one is not comfortable to use.

When I'm at my desk I plug into a 38" 4K monitor so I'd expect the GPU to drive an external 4k screen with decent performance; but I doubt that is a problem.

Context: I use my laptop for programming. As such screen real estate is critical. When on the 38" monitor, I can work on a center text editor; reference a tech doc on the left; run the compiler and a few status screens on the right. When mobile, I run 1080p with the text editor full screen and do a lot of alt-tab.

(off topic: I wish someone made a curved 34" monitor (17" * 2) my screen detail would remain constant for mobile and desk use.)