system76/laptop-suggestions #67: Cooling

Dmole, 2019-11-28:
### Why/User Benefit/User Problem
Heat and loud fans are no fun.

### Description of the feature
Put the CPU on the lid with passive cooling fins.

FiretronP75, 2019-11-28:
This is brilliant. I'd go a step further and say you don't even need storage under the keyboard. Just have an optional 2nd NVMe drive under the screen.

styr0, 2019-12-05:
Agreed. This is a good concept, which would likely require significant addition of thickness behind the built-in display (if you still call it a "lid" at that point). The heavy lid could be balanced out by some massive battery (or batteries) at the bottom part under the keyboard. Also, some built-in LED battery meters may be nice (related: [#69](https://github.com/system76/laptop-suggestions/issues/69)), but I could trade that for compatibility with other laptop batteries if feasible.

Having multiple external battery packs could also make them hot-swappable, without relying on harder-to-replace internal batteries (e.g., like some ThinkPads' internal batteries, which tend to have tragically short lifespans due to heat). Separating the battery area from the rest (CPU/GPU/chipsets/storage) in this way should greatly reduce battery heat. This in turn maintains battery capacity over the (prolonged) lifetimes of the batteries.

Putting the mobo/guts in the lid, and batteries in the base, all may add considerable thickness and weight. But I would be fine with something the size of a laptop from the '90s or '00s if the tradeoff yielded amazing performance and battery life. The last thing we need now is yet another MacBook knock-off (let's call them YAMBKOs for short).

The laptop industry over the past decade has been sacrificing thermal properties, performance, and battery life to make laptops thinner and lighter at all costs. It's still hard to find a decent laptop that will survive a long flight while doing real work on battery only, and without overheating itself into throttling hell. This trend needs to stop.

System76 already has a few ultraportables. A new laptop line built from scratch should have some modularity and at least the option of bulking up to improve cooling and to cram in more batteries to our hearts' desires. Sorry if this devolved into a rant ;)

styr0, 2019-12-05:
> ... Sorry if this devolved into a rant ;)

I totally missed #29 at the time of the above rant, which already goes to my points about batteries and thickness. But it is nonetheless worth considering the relationship between thickness and cooling, and better cooling helping batteries live longer.