system76/laptop-suggestions #68: LTE slot, single USB Type-C port and touchscreen

romanodev, 2019-11-28:
### Why/User Benefit/User Problem

Minimalist design, Developing on the go and annotations

### Description of the feature

An LTE slot would enable digital nomads to do their work on the go with no need for a wifi source. Ideally, the hardware should be compatible with Google Fi cards. 

Single Type-C port would allow a minimalist design. Currently, I am charging my Darter with the USB Type-C port (I didn't know it was possible! You need a >65 W adaptor) and I use a dock station for all other ports. Maybe System 76 can ship that too. 

The touch-enabled screen feature is for taking annotation while reading papers but also for having an iPad functionality within my laptop.

humanamerican, 2019-11-29:
This should be 3 separate issues so people can react/comment on them individually. For instance, I would like an LTE slot but am not interested in touch screen.

eddyp, 2019-12-05:
Agreed. These need to be 3 issues. For instance, only having USB-C would be a pain since you'd have to carry around some adapter whenever you need a mouse or connecting an external board or some other peripheral.

Dmole, 2019-12-05:
Charging via USB-C while a phone, mouse, UART are plugged in would be nice, as long as there is an adapter that's fine, but AFAIK there are no pure USB-C hubs.