system76/laptop-suggestions #69: Aesthetically pleasing LEDs for battery level on-button-press (PowerBook G4 like)

eddyp, 2019-11-29:
# User problem / Benefit

When on the go or when the laptop lid is sleeping, hibernating or off, sometimes you want to know how much charge you have.

The Apple PowerBook G4 batteries had a button and 5 beautiful 5 LEDs giving that rough info when pressing the button. Then they would turn off in about 2-3s.

With this you could have your laptop plugged in a power socket fur some juice until the last moment before the plane boarding or exiting the door, and you would know how much juice you had for the trip.

eddyp, 2019-11-29:
If you combine this with hot swappable external battery and having these LEDs on the battery itself, you have the benefit of being to charge your more depleted backup battery at first available power socket.

styr0, 2019-12-05:
Agreed. For reference, this is all I could find along these lines:
([Example PowerBook LED battery meter](http://www.notebookreview.com/assets/5163.jpg) from [TechTarget Notebook Review](http://www.notebookreview.com/notebookreview/apple-powerbook-review-pics-specs/))

I have some smaller handheld devices with battery meters like this. Some bigger ones may even have numerical (7-segment LED) displays to read out a percentage. Some external battery packs, e.g., for mobile phones, tablets, etc., work this way.

Some other devices I have can charge the batteries via the device's power supply (as you might expect, by plugging in a cable to power the device), but where the same batteries can also be charged outside the device by a charging port directly on the battery pack. That would also be a nice feature to accompany an LED battery meter--thus, you would not need to have the battery plugged into the laptop to charge it, and you could still see the charging progress.

Dmole, 2019-12-05:
I have a 2008? Aluminum MacBook with the battery led feature, it's nice, but one can just tap the keyboard brightness to turn on the screen to see the charge level, and tap again to turn the screen off.

eddyp, 2019-12-05:
> I have a 2008? Aluminum MacBook with the battery led feature, it's nice, but one can just tap the keyboard brightness to turn on the screen to see the charge level, and tap again to turn the screen off.

That does not address the "on the go" scenario  - imagine the laptop is already in the bag, you just slip your hand in the bag and press the button to see the charge level - and the "is the backup battery charged?" scenario.

Dmole, 2019-12-05:
Maybe I just never charge unless I'm already using the laptop... some replaceable batteries have feature on the battery itself which maybe more useful when deciding which one to swap out... but with 100wh, oled, and 7nm i imagine one would only need to charge while sleeping.