system76/laptop-suggestions #70: Sturdy AMD Development Laptop

audirs4, 2019-11-30:
| Development Laptop |  |
|---|---|
| Display Size | 14" |
| Display Resolution | 1920x1080 |
| CPU | Quad Core with SMT (AMD Ryzen would be preferred) |
| Memory | 2 Slots (not soldered) 8GB base |
| Storage | 2 storage options. 2 M.2 or 1 M.2 and 1 2.5" bay |
| Graphics | AMD CPU with integrated Vega |
| Keyboard | Backlit US. Good tactile feel |
| Power Adapter | USB Type C preferred |
| Ports | 2 USB type A ports |

**Size**
Need something portable, but it doesn't have to be crazy thin. I feel 14" is the perfect size. 15" laptops are just too bulky and 13" screens are just too small.

**Repairability**
I would rather have something a little thicker, but replaceable components, than everything soldered to the mainboard, making it very costly to upgrade or replace broken components.

**Battery Life**
Just need something that lasts over 6 hours or so doing normal tasks like web browsing to text editing or coding.

**HIDPI**
I feel high resolution screens are overkill and battery killers. What good is a super high resolution screen if you have to scale everything to make it readable. I feel at 14" that 1920x1080 is good, and perhaps 2480x1440 may work, or some variation of screen ratio in between. Doesn't have to be widescreen as movies will rarely be played.

**Graphics Card**
AMD APUs with Vega graphics will suite my needs. Definitely don't need dedicated (and definitely not Nvidia and their proprietary drivers). Would be used for the occasional lite gaming.

**Keyboard**
Backlit is a must, but a simple white is just fine. Do not need RGB. Keyboard keys needs to have a good tactile feel. I prefer a texture to super smooth.

**Touchpad**
Needs to feel smooth. Clicks must be easy. I typically do the tap to click when using the touchpad and not a mouse.

**Other**
Other than at least having 1 USB A for legacy reasons, everything else is optional. I would be fine using a dongle for HDMI or DisplayPort connections. RJ45 can be nice, but that won't affect my decision.

Dmole, 2019-12-05:
AMD has no modern laptop CPUs so you are going to be stuck with a battery draining heater.

audirs4, 2019-12-05:
> 
> 
> AMD has no modern laptop CPUs so you are going to be stuck with a noise, battery draining, heater.

Have you seen the latest Microsoft Surface Laptop 3 15"?

https://www.anandtech.com/show/15008/the-microsoft-surface-laptop-3-15-inch-review/8

Dmole, 2019-12-05:
@audirs4 It's not Zen 2 or 7nm; that 12nm Zen+ is not cool, and likely only available to M$.

audirs4, 2019-12-05:
> 
> 
> @audirs4 It's not Zen 2 or 7nm; that 12nm Zen+ is not cool, and likely only available to M$.

I never said it was Zen 2. You said I would be stuck with a noisy, battery draining heater. Here is a quote from the article I sent.

> The advantage of a larger laptop like the 15-inch Surface Laptop 3 is that it should not have much of an issue cooling a 15-Watt TDP, and the laptop was stress tested to find out if that was the case.
> 
> Interestingly the CPU on its own only pulls 9-10 Watts under sustained load, with a brief peak of a hair over 15 Watts. This is well under a typical Intel 15-Watt CPU which can pull well over 30 Watts for burst and sustained power draws can be over 15 Watts if the cooling system can handle it. Despite the lower than expected power draw under load the CPU is still able to maintain 3 Ghz or higher.
> 
> The cooling solution is very solid, and unless you are really working the laptop it stays silent for most of the time. Meanwhile if you do need to use everything the laptop has, the fan only gets up to 46 dB(A) measured one inch over the trackpad, which is not very loud. The tone of the air movement is a bit harsh though, but never gets to be a problem.

Not noisy. Not battery draining. Not a heater. But, as you stated, this is there 12nm Zen+ based CPU. Zen2 or even 3 will be out by the time System76 will be shipping these laptops. In which AMD will likely have the lead in the laptop space as they do currently in the desktop and server space.

Dmole, 2019-12-06:
Maybe it's perfectly awesome,
I'm just speculating because generally the lithography nm and architecture/process are the things that improve the heat to performance balance. 
12nm is way better than 65nm, (I have passive SBCs with both and the performance is about 67 times faster) Intel is only offering 10nm, but the 7nm from ARM or AMD server CPUs is the performance to power ratio that makes current laptop offering underwhelming.

anilsg, 2019-12-26:
Just want to support the 14" ideal size at 1920px. I'm looking for something to replace my Lenovo X1 Carbon and compete with Dell XPS

Owslla, 2020-01-02:
Showing support for the general build suggested above.