system76/laptop-suggestions #73: Bootable SD card slot

davidpaige, 2019-12-01:
I would like a bootable internal SD card slot, so I don't need an external USB key to boot into a new distro.

Dmole, 2019-12-05:
Why are you not using a bunch of partitions (practically unlimited with LVM/zfs/btrfs), or some VMs?

FiretronP75, 2019-12-05:
He might specifically want the other distro to be external, for added security, so it can be easily removed, so that way someone who steals your laptop doesn't have the secret drive in your pocket, unless they catch you and search you.

Dmole, 2019-12-05:
@FiretronP75 Full disk encryption and backups are the proper solution to the theft problem.

davidpaige, 2019-12-06:
> Why are you not using a bunch of partitions (practically unlimited with LVM/zfs/btrfs), or some VMs?

On my Acer laptop, I would burn the iso to an SD card, and boot from the internal card reader.  On my Lemur, I can't do that.  I need a USB key.

davidpaige, 2019-12-06:
> @FiretronP75 Full disk encryption and backups are the proper solution to the theft problem.

You are right, full disk encryption makes theft less of a concern.  It wasn't germane to this issue, though.

FiretronP75, 2019-12-06:
> @FiretronP75 Full disk encryption and backups are the proper solution to the theft problem.

Unless you fear they have your password, or will get it from you later. I know it is kinda extreme, but, there are people out there in extreme circumstances. An SD card is tiny and easily disposed of. People booting from external media for security is a real thing, not as uncommon as you might think. And an SD card would have advantages over a USB drive. It might not be David's concern. But, it is still valid for someone out there.

Dmole, 2019-12-06:
If you watch to consider extreme situations;

It's easier to put thermite on an internal drive than a removable SDCard.