system76/laptop-suggestions #75: 3:2 display options

sei6r, 2019-12-04:
### Why/User Benefit/User Problem
Laptops centred around work generally have little need for the media standard of 16:9; this turns vertical space into a precious commodity, and increases the need to scroll, as less information is being displayed at once. After all, nearly all UI is designed around adding components to the top and bottom of the screen (most common desktop environment defaults are a good example of this), leaving the user with a squished reading and writing experience. Manufacturers such as Microsoft, Google, and Huawei have already been using this aspect ratio in popular notebook lines for a while now.

### Description of the feature
Notebook display panels with resolutions such as 

- 1920 × 1280
- 2160 × 1440
- 3000 × 2000
- 3240 × 2160

 could all be used in upcoming laptops (those listed are only examples).

Dmole, 2019-12-05:
16:9 2550x1440 is ideal for 2 side-by-side windows (or one 4 window vimdiff 3-way merge)

FiretronP75, 2019-12-05:
Wide screen aspect ratio allows you to split your one wide screen into two tall virtual screens, using just one screen, saving you money, space, power, etc.

If you only need one tall window, just reduce the width of the window then center it. No need to ruin things for people who want to side-by-side tall windows without needing 2 screens.

loligans, 2019-12-09:
I would absolutely love a 3:2 or 16:10 aspect ratio screen. It makes for a better browsing experience in a web browser. It adds more vertical screen space which makes coding easier by being able to view more code. 

Most codebases exceeds the 80-character column limit which would make having 2 windows side-by-side pretty useless in my opinion. (Useless is subjective)

Dmole, 2019-12-09:
> Most codebases exceeds the 80-character column limit which would make having 2 windows side-by-side pretty useless in my opinion. (Useless is subjective)

2550x1440 (16:9) is 276 characters wide in a default Ubuntu terminal.; it's resolution/font/scaling not aspect ratio that determines width.

loligans, 2019-12-09:
> 2550x1440 (16:9) is 276 characters wide in a default Ubuntu terminal.; it's resolution/font/scaling not aspect ratio that determines width.

The fact of the matter is that screen resolution preference is subjective and could be debated all day long. As a consumer I would like to purchase a device with a 3:2 or 16:10 aspect ratio. I am willing to pay a premium for such device because the market lacks the variety I’m looking for. 

I’m looking for a 15” full aluminum body device with either 3:2 or 16:10 aspect ratio display.

mattwhiteside, 2019-12-10:
I think this is worth considering.  In other words, I don't think it's a given that 16:9 aspect ratio is better than 3:2 or 16:10.  16:9 is ideal for watching movies yes, but that's a small part of my usage.