system76/laptop-suggestions #76: Three-button trackpad

phma, 2019-12-05:
Can you make a trackpad with three buttons? That would make it easier to paste when the source or destination of the paste is a konsole. Three-button mice have been the norm for X for a long time.

FiretronP75, 2019-12-05:
Buttons are archaic. I think what you want is a 3 finger tap for the 3rd function, just like a 2 finger tap does a "right" click.

bryceadamprescott, 2019-12-08:
I (too) once hated button-less trackpads. Now, I like the 2 finger and 3 finger clicks more because they can be done anywhere on the trackpad. However, I still dislike taps; and no, a trackpad is not the same as a phone or tablet (this is what people say to me regarding this). I have to use a MacBook with macOS for work and having to use control+click, to right click, is so irritating.

phma, 2019-12-08:
If you need to click several times in the same place, you need buttons. This happens in CAD when you copy one line several times to the same place before rotating the copies by different angles, and when repeatedly using the same scroll button. If you try to do this with taps, the pointer moves slightly and messes up the drawing or moves off the button.

FiretronP75, 2019-12-09:
But when you need precision like that, you can't even use a trackpad to get on the right point to begin with before even doing the first click. You just have to use a mouse, preferable a trackball mouse so you can just lift your thumb after you get the right position, or get a mouse with a high DPI mode switch on it. Hmmm, maybe a trackpad could be useful for precision work if it has a high DPS mode switch....

eduncan911, 2019-12-26:
We're asking to drop the buttons all together, double the trackpad size, and modify the kernel to get rid of/reduce the hardcoded 'deadzones': https://github.com/system76/laptop-suggestions/issues/62

tenchiki, 2019-12-26:
I, too, prefer 3 physical buttons. Having the 3 buttons above the trackpad (like on some thinkpad models) is even better.

FiretronP75, 2020-01-03:
Hmm, that makes me wonder if it would be OK to have them left or right of the trackpad instead of above below, and maybe even change their format, like have them press like keyboard keys.