system76/laptop-suggestions #80: Allow every single light to be turned off

Thulinma, 2019-12-09:
### Why/User Benefit/User Problem
Bright lights really annoy me, especially when I am trying to sleep. Perhaps I am the only person that likes to have their laptop sitting in the same room they sleep in... or that often works in quite dark conditions.
I tend to "mod" all laptops I have so that the LEDs for power/HDD/etc are all very dim so that they do not disturb me by being the brightest thing in the room.

### Description of the feature
The ability to either control all LED's brightness from software (ideal) and/or to turn them all off with a toggle button or key combo (still quite nice) would be amazing. And (hopefully) quite simple to implement with no downsides.
If brightness is controllable it could even provide for some really cool booting/standby status feedback; but that's just for the shininess-factor, really, not the main reason to have this.

eduncan911, 2019-12-26:
This is what ticks me off with my Thinkpads.  yes, turn every light off.

arthurgleckler, 2020-01-08:
Yes, please.  My Lemur's power light is far too bright when I use it in the dark.

Thanks.