system76/laptop-suggestions #82: Smooth, Wide Range, Screen Brightness

FiretronP75, 2019-12-10:
### Why/User Benefit/User Problem
Bad quality screens don't get very bright at max. Good quality screens don't go very dark at min. And most if not all screens have a bit of a jump between the min and the step above min that feels like you jumped 3 or 4 steps between the 2 steps. And this is often times where I need the brightness to be, part way between the min and the first step.
### Description of the feature
A high maximum screen brightness. A very low minimum screen brightness. Many, perfectly even, steps between the min and max.

jonpolak, 2019-12-27:
Please ensure no PWM at low screen brightness!

FiretronP75, 2020-01-03:
Agreed! Pulse Width Modulation is much more important to consider. I'd rather be banned from dimming, than permitted to at the expense of my eyes/migraines.

But, if better dimming can be achieved without bad Pulse Width Modulation, I want it!

jonpolak, 2020-01-03:
@FiretronP75 or allow us to choose whether to sacrifice chromatic accuracy at low brightness but stay away from PWM.