system76/laptop-suggestions #86: Palm Rejection

bryanpaget, 2019-12-26:
### Why/User Benefit/User Problem

The palm rejection on the Darter Pro's trackpad is not very good.  It interferes with typing.

### Description of the issue

I can easily move the mouse and click with my palm from the corner of the trackpad.

The palm rejection works just as long as my palm is in the middle of the trackpad but fails when my palm is in any of the trackpad corners.

fbicknel, 2019-12-28:
I wish the trackpad were centered on the GH keys (or the spacebar). I think it might be out of the way from resting palms while typing. 

Unfortunately, I also wish the trackpad were larger, but then it would be impossible to rest your palms at all... unless, of course, we had a "clickable" track pad.... Can we have that? Pleeeeze?

bryanpaget, 2019-12-28:
Doesn't Apple have a patent on that?

https://appleinsider.com/articles/14/01/21/apples-new-trackpad-patent-replaces-click-button-with-force-sensors-adds-tactile-feedback

fbicknel, 2019-12-28:
Ugh. Of course they do. They're so annoying.
<https://about.me/fbicknel?promo=email_sig&utm_source=product&utm_medium=email_sig&utm_campaign=edit_panel&utm_content=thumb>
Frank Bicknell
about.me/fbicknel
<https://about.me/fbicknel?promo=email_sig&utm_source=product&utm_medium=email_sig&utm_campaign=edit_panel&utm_content=thumb>Schedule
time with me! <https://calendly.com/fbicknel>