system76/laptop-suggestions #89: Upgradable ram/hd/wifi

wontruefree, 2019-12-26:
### Why/User Benefit/User Problem
This lengths the lifecycle of the device and increases flexibility of the device after purchase.

### Description of the feature
have a back plate that can be removed and ram slots and hd slots available

DamianRivas, 2019-12-26:
Yes!! Please don't follow Dell's steps with their XPS. Lack of upgradability, and being unable to fix the WiFi card that came faulty, were the reasons I returned my Dell and gave System 76 a chance, a choice I never regretted!

I'm sure Linux users care more about upgradability and performance than "sleek and thin". The Galago Pro balances the two just perfectly.

MikeStutzzzz, 2019-12-26:
I agree being able to upgrade or replace a single component is nice. I would also like the ability to purchase the laptops themselves without Drives and RAM. If I upgraded my Previous model to 32 Gigs of RAM and a fresh M.2 Samsung Drive, its a hassle to order a new laptop and try to sell the old parts.