system76/laptop-suggestions #99: Keyboard, trackpad, and screen sourcing

marcstein, 2019-12-26:
The very best keyboard, trackpad and screen that I've yet seen on a laptop all are to be found in the Google Pixel reference laptop. If fact, I'm using both a 1st gen and 2nd gen version as linux machines still.

From an ergonomic standpoint these are much better than those elements on MacBook Pros or high end PC laptops. If you could produce an i9 32 gb laptop with those components with reasonable thermal transfer and an nvidia GPU it would be truly amazing!

jonpolak, 2019-12-27:
Would be great if you could give us stats on the screen like whether it uses PWM or not. High end components would be great!!

marcstein, 2019-12-27:
Hi Jonathan,

The base specs are 12.85-inch high resolution IPS (2560 x 1700, at 239 PPI)
400 nit, 3:2 aspect ratio, 178° viewing angle. Not sure on PWM or
frequency, but the response is very fast and I'm quite sensitive to screen
artifacts and I like the screen much better than a Retina display.

The *12.85-inch* *glossy *touchscreen is the highlight of the hardware that
delivers an experience unlike any other traditional offering. The size and
aspect ratio are certainly not commonplace for a notebook, but no one is
going to argue against a display with *over twice the pixels of a fullHD
screen *and an even higher DPI than Apple’s current crop of Retina
MacBooks. The screen even appears to be plane to the display bezel for a
greater "pop-up" effect instead of being buried under its Gorilla Glass
layer. Texts are fortunately scaled to more legible sizes by default and
can be zoomed out quite far if desired without losing visual quality. It’s
also important to note that even though the screen size is small, its
aspect ratio and high horizontal resolution make transitioning from 720p or
1080p displays to the unusual screen size of the pixel easier than
initially expected.
[image: Notebookcheck on native 1920 x 1200 resolution]
<https://www.notebookcheck.net/fileadmin/_migrated/pics/1080p_03.png>Notebookcheck
on native 1920 x 1200 resolution
[image: Same page as displayed on the 2560 x 1700 resolution Pixel]
<https://www.notebookcheck.net/fileadmin/_processed_/csm_Screenshot_2013-04-14_at_1.02.14_AM_02_85fe160e46.png>Same
page as displayed on the 2560 x 1700 resolution Pixel
[image: Same page as displayed on the Pixel at 50% font size]
<https://www.notebookcheck.net/fileadmin/_processed_/csm_Screenshot_2013-04-14_at_1.02.54_AM_02_44009a0bda.png>Same
page as displayed on the Pixel at 50% font size

Measured *brightness *averaged about *365 nits* over nine quadrants, which
is not quite up to the 400 nit claim by Google, though still more than
sufficient for indoor use and most outdoor conditions. The *very low black
level*, measured on a completely dark screen on maximum brightness, paves
way for a very high contrast ratio that surmounts most of the competition.
Subjectively, it is certainly one of the highest quality displays currently
available in the notebook market.
361
cd/m² 367
cd/m² 362
cd/m²
361
cd/m² 376
cd/m² 375
cd/m²
337
cd/m² 351
cd/m² 380
cd/m²
*Distribution of brightness*
LG LP129QE1-SPA1
Gossen Mavo-Monitor
Maximum: 380 cd/m² Average: 363.3 cd/m²
Brightness Distribution: 89 %
Center on Battery: 376 cd/m²
Contrast: 1106:1 (Black: 0.34 cd/m²)

According to measurements from the X-Rite i1 Basic Pro 2 spectral
profiler,* color
temperature* and *gamma *are almost perfectly at* 6500K* and *2.2*,
respectively, which are in line with both D65 and sRGB standards. Colors
are generally represented more accurately the more saturated they become
with the exception of blue. However, this may simply be due to the cooler
color temperature of the screen as all other colors fall well below the 10
percent DeltaE 2000 deviation threshold.
[image: Grayscale]
<https://www.notebookcheck.net/fileadmin/_migrated/pics/pixelgrayscale_02.png>
Grayscale
[image: Color accuracy]
<https://www.notebookcheck.net/fileadmin/_migrated/pics/pixelcolor_02.png>Color
accuracy
[image: Color saturation]
<https://www.notebookcheck.net/fileadmin/_migrated/pics/pixelsat_02.png>Color
saturation
[image: Use under direct sunlight is possible at maximum brightness]
<https://www.notebookcheck.net/fileadmin/_processed_/csm_IMAG0448_02_bc206c1ecc.jpg>Use
under direct sunlight is possible at maximum brightness

*Outdoor usability* is *good *provided that distracting glare can be
reduced or avoided. Luckily, the display allows for a very wide range of
viewing angles, so users can adjust the display to almost any position
permissible by the hinges and still be able to see onscreen content. Sunny
days with bright overcasts will of course demand *higher brightness
settings. *Still, viewability is very good at the cost of some glare and
lower battery life.
[image: Viewing angles Google Chromebook Pixel]
<https://www.notebookcheck.net/fileadmin/_processed_/csm_IMAG0432_02_8353e8777b.jpg>Viewing
angles Google Chromebook Pixel

*Viewing angle stability* is *excellent *due to the underlying *IPS *panel.
Colors and brightness do not radically shift even if viewing from extreme
angles, so sharing the display with adjacent viewers is not problematic.

Marc Stein CEO
underwrite.ai
P. (617) 545-7171   M. (857) 222-3291
E. marc.stein@underwrite.ai
W. https://www.underwrite.ai
<https://www.linkedin.com/in/marc-stein-b83b765>

jonpolak, 2019-12-27:
Those are all good stats, I don't disagree. What I'm asking about has to do with dimming. What's the process by which dimming is accomplished? If it uses PWM (pulse width modulation) to regulate the ON/OFF duty cycle for the LED backlight then that creates flicker. Many good screens don't use that, instead they use a regulated DC current source. You may be able to detect flicker using your cell phone camera. The flicker gives many users headaches after a few hours of use

marcstein, 2019-12-27:
See if this helps. Here is a blur test at high brightness and low
brightness and a shot of the desktop. Sorry for the crappy pics.

high
[image: high_IMG_20191227_143147.jpg]

low

[image: low_IMG_20191227_143539.jpg]

desktop

[image: desktop_IMG_20191227_142957.jpg]
Marc Stein CEO
underwrite.ai
P. (617) 545-7171   M. (857) 222-3291
E. marc.stein@underwrite.ai
W. https://www.underwrite.ai
<https://www.linkedin.com/in/marc-stein-b83b765>

jonpolak, 2019-12-27:
the refresh rate of the screen and the PWM switching frequency are not related. Here's the complete list of [every machine ever reviewed by Notebookcheck](https://www.notebookcheck.net/PWM-Ranking-Notebooks-Smartphones-and-Tablets-with-PWM.163979.0.html)  you'll see lots of very high-end devices with PWM of 0 (scroll to the bottom). As well as some > 100kHz (100,000 Hz). The issue is flicker not refresh rate.