system76/laptop-suggestions #102: No PWM for screen dimming

jonpolak, 2019-12-27:
### Why/User Benefit/User Problem
Pulse Width Modulation PWM is often used to control the screen brightness by reducing the duty cycle of the LED backlight. This is problematic because it is actually flashing very quickly instead of uniform light. This gives many users headaches. 
### Description of the feature
Use DC current regulator to control the brightness of the backlight

mmstick, 2019-12-27:
The brightness of a diode is not controlled by the amount of current given to it, hence why LED light bulbs in a dimmable switch are either completely off, or complete on. Brightness of a LED can only be simulated using PWM, and dimmable LED bulbs often use a chip to convert current differences into PWM. The frequency of the PWM shifts is so high that your eyes can't tell the difference. So if you're getting headaches, it's because the frequency of the shifting is far too low.

jonpolak, 2019-12-27:
Hi Mike -- LED brightness can and is often controlled with analog (current driven) modes. [Here's a 2011 application note from Aimtec](http://www.aimtec.com/site/Aimtec/files/documents/ApplicationNotes/a031e%20pwm%20vs%20analog%20dimming%20of%20leds.pdf)

Here's just the chart form digikey's explanation of rationale from PWM back in 2016 [Luminous Flux vs Current](https://www.digikey.ca/-/media/Images/Article%20Library/TechZone%20Articles/2016/October/How%20to%20Dim%20an%20LED%20Without%20Compromising%20Light%20Quality/article-2016october-how-to-dim-an-led_fig1.jpg). You can read the [article here](https://www.digikey.ca/en/articles/techzone/2016/oct/how-to-dim-an-led-without-compromising-light-quality): 

PWM became the standard in cheap displays because it's the easiest way to maintain chromacity stability with current changes. [Notebook Check](https://www.notebookcheck.com) always checks the PWM characteristics of displays. Those with low freq switching will give you a terrible headache.