system76/laptop-suggestions #116: Provide a version of ffmpeg with nvenc encoder support out of the box.

Shickmo, 2020-01-05:
### Why/User Benefit/User Problem
Currently Pop!_OS doesn't ship with a version of ffmpeg that is complied with nvenc encoder support out of the box. 

### Description of the feature
Okay so this one is a bit of a stretch but hear me out. Since creators are one of the audiences you are targeting this hardware to and Nvidia GPU's are still the lion share of what is being produced, and likely still will be by the time these devices would come out, this seems like a fairly glaring omission to be rectified. 

Now technically you can install a snap of ffmpeg, that Martin Wimpress put together, but why should that need to be a step taken on hardware targeted for creators / makers and the like? If smaller shops like Elementary, still based on Ubuntu 18.04 to be fair, and Solus can provide this out of the box experience I am confident that System76 can as well.

curiousercreative, 2022-07-15:
This is a software request. Moreover, the handbrake flatpak now comes with NVENC support and likely fits most needs.

brs17, 2023-01-20:
General Pop!_OS requests/issues should be brought up here: https://github.com/pop-os/pop/issues

closed 2023-01-20 by brs17