system76/laptop-suggestions #118: Single cable docking solution for this laptop.

Shickmo, 2020-01-05:
### Why/User Benefit/User Problem
There is no docking solution I am aware of that could power an Oryx Pro level device with a one cable connection.

### Description of the feature
This is a bit of an off shoot of a previous issue I put in but will highlight the same HP dock to provide some inspiration for you, https://store.hp.com/us/en/pdp/hp-thunderbolt-dock-g2-with-combo-cable. If you could come up with a device like this for the new laptop/s it would be pretty game changing for the open source hardware market.

DylanRob, 2020-08-08:
Something like this may help you: https://www.amazon.com/Plugable-Charging-Delivery-Specific-Thunderbolt/dp/B01FKTZLBS/ref=sr_1_7?s=electronics&ie=UTF8&qid=1501702775&sr=1-7&keywords=usb+c+dock looks like it should work on the Oryx Pro as it seems to be roughly the same as a Clevo P650SE from that i have seem from the internet : http://forum.notebookreview.com/threads/p650-and-usb-c-dock-with-video.807579/

timnolte, 2020-12-21:
@ghost the Oryx Pro can't be powered via Thunderbolt from what I can tell.