system76/laptop-suggestions #121: Azerty layout for laptops

gsouf, 2020-01-09:
### Why/User Benefit/User Problem

In France you'll find that maybe 99% or so of keyboards use AZERTY layout. For the most it's a deal breaker to have QWERTY layout when buying a laptop. 

### Description of the feature

Have an option for AZERTY keyboards when configuring a laptop.

GroobleDierne, 2022-05-20:
It's absolutely needed! And it would be a fun occupation for you keyboards nerds 😁

CircleCode, 2025-08-02:
Hi, sorry to dig up, but is there any progress on this? I would really love a system76, but without azerty keyboard, it is complicated… Maybe just a keyboard cover coul be enough (just thinking out loud), even if a true AZERTY keyboard would be a must