system76/laptop-suggestions #123: Mechanical volume control

ninernet, 2020-01-30:
Thank-you for soliciting these feature requests. This is unprecedented in my experience.

### Why/User Benefit/User Problem
The problems is that, without a mechanical volume control -- like the traditional, edgeways-oriented rotating knob -- there is a lot of clicking and dragging involved. My definition of "a lot" is anything more than zero clicks or use of a mouse, keyboard and/or screen at all. Use of a hardware, mechanical volume control is as instant as my wetware reflexes.

### Description of the feature
I've already mentioned the traditional, edgeways-oriented rotating knob, but any type of mechanical control that makes sense on a laptop would be good.

ninernet, 2020-02-04:
I'm bumping this only because Github "flagged" my account after I posted it, so I'm not sure whether or not that affected this issue.

Dmole, 2020-06-06:
Mechanical things are often of the cheap breaky flavor, so one might consider minimal io an advantage.

You could buy a USB DAC for that if using headphones.

ninernet, 2020-06-06:
Thanks for your suggestion, but if one is using headphones there is usually volume control built into them. I'm clearly referring to controlling the built-in speakers.

If I understand your use of the term "minimal io" correctly, that's what I'm calling for. Reaching for a physical volume knob is way more minimal than fiddling with a mouse and/or keyboard. If you're referring to minimal/fewer physical inputs, along with your reference to cheap and "breaky", well, the solution to that is obvious: don't use cheap and breaky parts.