system76/laptop-suggestions #124: A nice budget build

KleinKravis, 2020-02-06:
You guys could offer a 11-12” laptop with a Ryzen 3 or a Athlon and 2-4 GBs of RAM. I would definitely buy that.

Dmole, 2020-06-06:
8GB RAM is really a minimum these days with heavy apps like firefox and LibreOffice, even raspberrypi has 8 for $75. There was that $200 3rd gen AMD from Walmart briefly. https://www.phoronix.com/scan.php?page=article&item=ryzen3-windows-linux&num=1

KleinKravis, 2020-06-23:
I've been running Gimp or Gravit plus Chrome with KDE on 1.2ghz ARM Cortex and 1gb of RAM (936MB Usable) I think 4 is fine

closed 2021-10-14 by anisestar

anisestar, 2021-10-14:
Ah okay, I was just closing old issues

leviport, 2021-10-14:
No worries, and thanks for the suggestion!

jadeastar, 2022-06-04:
Just passing through to propose an update, the original but with 8GiB RAM and some kind of Arm chip