system76/laptop-suggestions #128: AMD Renoir (4xxx) (Zen 2)

GreyXor, 2020-04-09:
### Why/User Benefit/User Problem
Notebooks with AMD 4000 series processors APU

### Description of the feature
Great energy savings, great power. Much better than their Intel equivalent.
https://en.wikichip.org/wiki/amd/cores/renoir

eduncan911, 2020-04-22:
Please this.  It also shaves off a hundred or two dollars.

I hate paying the Intel tax, as well as the backdoor taxes.  Even with Coreboot, Intel still requires proprietary blobs loaded.

briggs1978, 2020-04-27:
I have to imagine it's already on the horizon, but very much this.

eduncan911, 2020-04-27:
I've literally been asking System76 Twitter for almost a year:

https://mobile.twitter.com/search?q=%40eduncan911%20system76%20amd&src=typed_query

Dmole, 2020-06-02:
The competition is looking tempting

https://www.phoronix.com/scan.php?page=article&item=amd-ryzen5-4500u&num=1

eduncan911, 2020-06-06:
> The competition is looking tempting
> 
> https://www.phoronix.com/scan.php?page=article&item=amd-ryzen5-4500u&num=1

Yes, especially since they can be ordered with Linux now and not paying the Microsoft tax.

Coreboot is also available for many variants of AMD, so there had been partnerships between them.  Now rather Renoir is included in those specs I do not know.

https://doc.coreboot.org/search.html?q=Amd&check_keywords=yes&area=default

Excluding the coreboot "feature", which can be implemented later with a BIOS update, there's no reason not to offer AMD.

GreyXor, 2020-07-29:
https://twitter.com/jeremy_soller/status/1286457590289858560?s=20

eduncan911, 2021-01-14:
Well, the Pangolin has been announced!

https://system76.com/laptops/pangolin

But still 1080p.  Dammit!!!

Pop_OS! has been the best 4k distro I've ever used (on non-System76 machines).  Perfection.  But, no 4k available to buy?!?!  Argh...

GreyXor, 2021-01-14:
> Well, the Pangolin has been announced! **Guess we can close this thread now**. :)
> 
> https://system76.com/laptops/pangolin
> 
> But still 1080p. Dammit!!!
> 
> Pop_OS! has been the best 4k distro I've ever used (on non-System76 machines). Perfection. But, no 4k available to buy?!?! Argh...

Yass!
see you people : https://github.com/system76/laptop-suggestions/issues/143

closed 2021-01-14 by GreyXor

dylangerdaly, 2021-03-18:
Looks like everything is closed tho, UEFI + EC is all closed :-1: